CadQuery/cadquery #1527: cadquery and vtk compatibility

msavtchouk-pf, 2024-03-01:
Hi all
We are using the latest pip-installable cadquery and we also recently added a new dependency (vtk==9.3.0)
Just importing those two in the same file:
 `import cadquery
import vkt 
`
Results in the following error:
`ImportError: Initialization failed for vtkWebCore, not compatible with vtkmodules.vtkCommonCore`

Since that the vtk version, used by cadquery is incopartible with vtk==9.3.0.
I could not find any pointers how to resolve this so far. Any help would be appreciated!

msavtchouk-pf, 2024-03-01:
Update: it is possible that our build system is not resolving the versions correctly. I am waiting for response from them to confirm that.

jmwright, 2024-03-01:
Could be related to [this](https://github.com/CadQuery/ocp-build-system/issues/14#issuecomment-1531252029). I think that eventually we will remove VTK from the OCP wheels and have it as a dependency, but it is bundled for now. If conda is an option in your environment, you could try using that to make sure that there is not a deeper incompatibility that has been introduced in VTK 9.3.0 [reference](https://github.com/CadQuery/cadquery/blob/master/environment.yml#L8C2-L8C14).