CadQuery/cadquery #1532: Work around lacking robustness of isEqual() and isSame() functions

mxmws, 2024-03-09:
Hi,

extremely small changes due to rounding already result in `isEqual()` and `isSame()` returning false. For example when I export a shape as a `.step` file and import it again it's not considered the same anymore.

I'm hoping for suggestions on how to work around this! One idea was to compare parameters like area and center of mass and consider two shapes equal if the difference between those parameters is very small. If someone has a better idea please suggest it to me. Thanks!

adam-urbanczyk, 2024-03-09:
You are misinterpreting what those methods are for, see the docstring:
```
Returns True if other and this shape are same, i.e. if they share the
same TShape with the same Locations. Orientations may differ.
```
They are checking the internal representations of the Shape objects, not geometric equality.

If you want geometric approximate equality, you'll have to implement it yourself. You could indeed compare center of mass, volume, area, number of vertices/edges/faces. I do wonder why do you actually need it.

mxmws, 2024-03-09:
Thank you for your reply. I need it to evaluate a library that generates CSG trees from step files. So I'm converting step to CSG and back to step to see if it's still the same.

adam-urbanczyk, 2024-03-09:
If you are able to load the model back to CQ, you could use bool cut to compare the original and final. There is a `tol` parameter that might be very useful in your use case.

mxmws, 2024-03-10:
You mean I could use a fuzzy boolean operation to subtract one shape from the other and see if anything is left? That's a cool idea! Thank you