CadQuery/cadquery #1535: Hole importing to 3D slicer as a separate object

cgn1234, 2024-03-12:
I'm trying to create an object with holes in it for magnets. I'm creating two disk shaped holes for magnets in the middle of the object using the same code for each. One of the holes seems fine.

With the other one, when I bring the object into the Slicer and try to slice, it complains that there are two overlapping objects and won't slice.

I have two objects in the output, each with a magnet in them. In the slicer, when I split the object, so I get two separate objects that I can move around on the bed, I end up with a 3rd object which is the solid disk. 

I've changed the code to generate the hole to a number of different approaches:

Building it with circle, extrude(combine="s")
Creating a separate object and cutting it out of my first object
Using the hole method

All generate the same results. 

I tried making the size of the hole wider so it sticks out of the object to verify that it's in the right place, which it is.

Any ideas? 

Thanks,
/Chris

jmwright, 2024-03-13:
Which format are you expecting to? If you're not using STL, try that. Otherwise, can you share a minimal example that reproduces the issue?

cgn1234, 2024-03-14:
Thanks for responding. 

I've stopped having the issue, no idea what caused it. :-(

/Chris

closed 2024-03-14 by cgn1234