CadQuery/cadquery #1538: [need help] A few questions on the door assembly's constraints in the docs' Assembly tutorial section

huskier, 2024-03-16:
Hi, I do not quite understand the following several constraints extracted from the door assembly example. Could any one explain them a little bit? Thanks in advance.
```
    # top
    .constrain("top@faces@<Z", "con_tl?X", "Plane")

    # bottom
    .constrain("bottom@faces@>Z", "con_bl?X", "Plane")

    # right connectors
    .constrain("top@faces@>Z", "con_tr@faces@>X", "Plane")
    .constrain("bottom@faces@<Z", "con_br@faces@>X", "Plane")
```

BTW, I think the panel constraints may need one more in Y-direction. Otherwise, I think the panel is not fully constrained.
```
    # panel
    .constrain("left@faces@>X[-4]", "panel@faces@<X", "Plane")
    .constrain("left@faces@>Z", "panel@faces@>Z", "Axis")
```

lorenzncode, 2024-03-17:
`.constrain("top@faces@<Z", "con_tl?X", "Plane")`

is similar to:

`.constrain("top", <the top>.faces("<Z").val(), "con_tl", <the con_tl>.faces(tag="X").val(), "Plane")`

The "?" is used with tags.  "con_tl?X" refers to the tag named "X" on connector part "con_tl".

This line adds a Plane constraint between the face selected on the top part and the face tagged with tag name "X" on the con_tl connector part.

>there is a flaw (see the following screenshot) for the door display with the show function imported from cadquery.vis,

I reproduced this display issue and opened PR #1540.

huskier, 2024-03-18:
Thank you for your reply and the PR, lorenzncode. @lorenzncode 

> This line adds a Plane constraint between the face selected on the top part and the face tagged with tag name "X" on the con_tl connector part.

My question is exactly here. The face selected on the top part is perpendicular to Z-axis, while the face tagged with tage name "X" on the con_tl connector part is perpendicular to X-axis. So, how to understand this Plane constraint? It is not the touch relationship.

lorenzncode, 2024-03-19:
>The face selected on the top part is perpendicular to Z-axis, while the face tagged with tage name "X" on the con_tl connector part is perpendicular to X-axis. So, how to understand this Plane constraint?

Correct, initially these two faces are in different planes.  The desired end result after `solve` is that the two faces are in the same plane.

closed 2024-03-19 by jmwright

huskier, 2024-03-22:
I still do not quite understand the constraint. Let me explain my point in detail.

```
    # top
    .constrain("top@faces@<Z", "con_tl?X", "Plane")
```
For the above code, "top@faces@<Z" indicates the  plane XY-plane002.
[image]

while "con_tl?X" indicates the plane far most in X-direction parallel to the plane YZ-plane002
[image]

The  plane XY-plane002 and the plane far most in X-direction parallel to the plane YZ-plane002 are orthogonal to each other. I think these two planes never touch each other.

lorenzncode, 2024-03-23:
See the docs on constraints and the constraint solver here:
https://cadquery.readthedocs.io/en/latest/assy.html#constraints

When you call `solve` it runs an optimization that iterates by automatically varying the component locations.  The optimization attempts to satisfy all constraints.  

Prior to optimization, it is quite typical for constraints to not be met.  If all constraints were already met prior to optimization then there wouldn't be any reason to run the optimization.

Before optimization: 

The top face and con_tl face are in different planes as shown in red, green.

[image]

After optimization:

The Plane constraint is satisfied.

[image]

huskier, 2024-03-24:
@lorenzncode Thank you for your detailed explanation. I've gotten the point.

To simplified the constraints setting, I've firstly put all the parts in a pre-arranged layout, and then set the constraints in the new initial layout. I've opened a new thread (#1546) for the approach.

PR #1540 (2024-03-19): Fix VTK assembly render apply gp_Intrinsic_ZXY rotation
Resolves #1538 

See also `SetOrientation` https://vtk.org/doc/nightly/html/classvtkProp3D.html

>   Sets the orientation of the Prop3D.
> 
>   Orientation is specified as X,Y and Z rotations in that order, but they are performed as RotateZ, RotateX, and finally RotateY.

`Location` might instead be enhanced to provide optional parameter to set `gp_EulerSequence` say in `toTuple`.  Currently hardcoding in the VTK function.