CadQuery/cadquery #1542: How to create an array of Workplane objects

GaN-T, 2024-03-22:
I am trying to create an array from an imported model (.step). How do I do this?

```
base = cq.importers.importStep("myModel.step")
rarray = cq.Workplane("front").rarray(19,19,3,3,False)   # points on which I would like the copies of the model to be centered on
```

GaN-T, 2024-03-22:
my current solution is as follows. But is there a more direct way?

```
merged = base
for loc in rarray.vals():
    merged += base.translate(loc)
```

michaelgale, 2024-03-22:
This alternative also works:

```python
merged = (
    cq.Workplane("front")
    .rarray(19, 19, 3, 3, False)
    .eachpoint(lambda loc: base.val().located(loc), combine=True)
)
```

Your solution is also valid likely the underlying geometric representation is the same, i.e. the union of multiple solids into a composite compound.

closed 2024-03-22 by GaN-T