CadQuery/cadquery #1543: For a complete new python venv environment, there is no ipython dependency installation through the command "pip install cadquery".

huskier, 2024-03-22:
For a complete new python venv environment, there is no ipython dependency installation through the command "pip install cadquery". I have to manully install ipython to make some codes work.

I've looked the file of setup.py, "ipython" is in the extras_require list. 
```
    extras_require={
        "dev": [
            "docutils",
            "ipython",
            "pytest",
            "black@git+https://github.com/cadquery/black.git@cq",
        ],
        "ipython": ["ipython",],
    },
```
Another package dependent question is that where the vtkmodules dependency is explicitly written in the file of setup.py to support cadquery.vis?

adam-urbanczyk, 2024-03-24:
cadquery-ocp provides vtk. There is no hard dep on ipython AFAIK.

huskier, 2024-03-24:
> cadquery-ocp provides vtk.

Thanks, I've gotten it.

The ipython dependency is introduced by vis.py. The following is the traceback.
> (cpvenv) (base) E:\software\hobbies\CAD\CadQuery\tests>python AxisConstraints.py
> Traceback (most recent call last):
>   File "E:\software\hobbies\CAD\CadQuery\tests\AxisConstraints.py", line 3, in <module>
>     from cadquery.vis import show
>   File "E:\software\hobbies\CAD\cpvenv\Lib\site-packages\cadquery\vis.py", line 3, in <module>
>     from .occ_impl.jupyter_tools import DEFAULT_COLOR
>   File "E:\software\hobbies\CAD\cpvenv\Lib\site-packages\cadquery\occ_impl\jupyter_tools.py", line 4, in <module>
>     from IPython.display import Javascript
> ModuleNotFoundError: No module named 'IPython'

adam-urbanczyk, 2024-03-24:
FYI `vis` is for now experimental and meant for devs. It should not depend on ipython, so opening an issue.

closed 2024-03-24 by adam-urbanczyk