CadQuery/cadquery #1550: segfault during extrusion test

greyltc, 2024-03-26:
## To Reproduce
The error is here on the 2.4.0 release and on 4f2cec0f5ee21041a1f8c80f161334f349180933 with the OCP 7.7.2.0 release.

```bash
python -m pytest tests
```

## Backtrace

```
platform linux -- Python 3.11.8, pytest-8.1.1, pluggy-1.4.0
PySide6 6.6.2 -- Qt runtime 6.6.2 -- Qt compiled 6.6.2
rootdir: REDACTED/python-cadquery/src/cadquery
plugins: pylama-7.7.1, flaky-3.7.0, ordering-0.6, xvfb-2.0.0, cov-4.1.0, mock-3.12.0, anyio-4.3.0, forked-1.6.0, qt-4.4.0, rerunfailures-14.0, dash-2.15.0, asyncio-0.23.6, hypothesis-6.99.11
asyncio: mode=Mode.STRICT
collected 540 items / 18 deselected / 522 selected                                                                                                                                                                                                                                                                     

tests/test_assembly.py .........................................................................                                                                                                                                                                                                                 [ 13%]
tests/test_cad_objects.py .....................................                                                                                                                                                                                                                                                  [ 21%]
tests/test_cadquery.py ...................................................Fatal Python error: Segmentation fault

Current thread 0x0000780a36619740 (most recent call first):
  File "REDACTED/python-cadquery/src/cadquery/cadquery/occ_impl/shapes.py", line 385 in clean
  File "REDACTED/python-cadquery/src/cadquery/cadquery/cq.py", line 3278 in <listcomp>
  File "REDACTED/python-cadquery/src/cadquery/cadquery/cq.py", line 3277 in _combineWithBase
  File "REDACTED/python-cadquery/src/cadquery/cadquery/cq.py", line 3127 in extrude
  File "REDACTED/python-cadquery/src/cadquery/tests/test_cadquery.py", line 3619 in testExtrude
  File "/usr/lib/python3.11/unittest/case.py", line 579 in _callTestMethod
  File "/usr/lib/python3.11/unittest/case.py", line 623 in run
  File "/usr/lib/python3.11/unittest/case.py", line 678 in __call__
  File "/usr/lib/python3.11/site-packages/_pytest/unittest.py", line 321 in runtest
  File "/usr/lib/python3.11/site-packages/_pytest/runner.py", line 172 in pytest_runtest_call
  File "/usr/lib/python3.11/site-packages/pluggy/_callers.py", line 102 in _multicall
  File "/usr/lib/python3.11/site-packages/pluggy/_manager.py", line 119 in _hookexec
  File "/usr/lib/python3.11/site-packages/pluggy/_hooks.py", line 501 in __call__
  File "/usr/lib/python3.11/site-packages/flaky/flaky_pytest_plugin.py", line 146 in <lambda>
  File "/usr/lib/python3.11/site-packages/_pytest/runner.py", line 340 in from_call
  File "/usr/lib/python3.11/site-packages/flaky/flaky_pytest_plugin.py", line 145 in _call_runtest_hook
  File "/usr/lib/python3.11/site-packages/flaky/flaky_pytest_plugin.py", line 149 in call_and_report
  File "/usr/lib/python3.11/site-packages/_pytest/runner.py", line 134 in runtestprotocol
  File "/usr/lib/python3.11/site-packages/_pytest/runner.py", line 115 in pytest_runtest_protocol
  File "/usr/lib/python3.11/site-packages/flaky/flaky_pytest_plugin.py", line 89 in pytest_runtest_protocol
  File "/usr/lib/python3.11/site-packages/pluggy/_callers.py", line 102 in _multicall
  File "/usr/lib/python3.11/site-packages/pluggy/_manager.py", line 119 in _hookexec
  File "/usr/lib/python3.11/site-packages/pluggy/_hooks.py", line 501 in __call__
  File "/usr/lib/python3.11/site-packages/_pytest/main.py", line 364 in pytest_runtestloop
  File "/usr/lib/python3.11/site-packages/pluggy/_callers.py", line 102 in _multicall
  File "/usr/lib/python3.11/site-packages/pluggy/_manager.py", line 119 in _hookexec
  File "/usr/lib/python3.11/site-packages/pluggy/_hooks.py", line 501 in __call__
  File "/usr/lib/python3.11/site-packages/_pytest/main.py", line 339 in _main
  File "/usr/lib/python3.11/site-packages/_pytest/main.py", line 285 in wrap_session
  File "/usr/lib/python3.11/site-packages/_pytest/main.py", line 332 in pytest_cmdline_main
  File "/usr/lib/python3.11/site-packages/pluggy/_callers.py", line 102 in _multicall
  File "/usr/lib/python3.11/site-packages/pluggy/_manager.py", line 119 in _hookexec
  File "/usr/lib/python3.11/site-packages/pluggy/_hooks.py", line 501 in __call__
  File "/usr/lib/python3.11/site-packages/_pytest/config/__init__.py", line 174 in main
  File "/usr/lib/python3.11/site-packages/_pytest/config/__init__.py", line 197 in console_main
  File "/usr/lib/python3.11/site-packages/pytest/__main__.py", line 7 in <module>
  File "<frozen runpy>", line 88 in _run_code
  File "<frozen runpy>", line 198 in _run_module_as_main

Extension modules: markupsafe._speedups, zmq.backend.cython.context, zmq.backend.cython.message, zmq.backend.cython.socket, zmq.backend.cython._device, zmq.backend.cython._poll, zmq.backend.cython._proxy_steerable, zmq.backend.cython._version, zmq.backend.cython.error, zmq.backend.cython.utils, tornado.speedups, psutil._psutil_linux, psutil._psutil_posix, _cffi_backend, simplejson._speedups, xxsubtype, shiboken6.Shiboken, PySide6.QtCore, PySide6.QtGui, PySide6.QtWidgets, PySide6.QtTest, vtkmodules.vtkCommonCore, vtkmodules.vtkCommonMath, vtkmodules.vtkCommonTransforms, vtkmodules.vtkCommonDataModel, vtkmodules.vtkCommonExecutionModel, vtkmodules.vtkCommonMisc, vtkmodules.vtkFiltersCore, numpy.core._multiarray_umath, numpy.core._multiarray_tests, numpy.linalg._umath_linalg, numpy.fft._pocketfft_internal, numpy.random._common, numpy.random.bit_generator, numpy.random._bounded_integers, numpy.random._mt19937, numpy.random.mtrand, numpy.random._philox, numpy.random._pcg64, numpy.random._sfc64, numpy.random._generator, matplotlib._c_internal_utils, PIL._imaging, matplotlib._path, kiwisolver._cext, ezdxf.acc.vector, ezdxf.acc.matrix44, ezdxf.acc.construct, ezdxf.acc.bezier4p, ezdxf.acc.bezier3p, ezdxf.acc.bspline, pandas._libs.tslibs.np_datetime, pandas._libs.tslibs.dtypes, pandas._libs.tslibs.base, pandas._libs.tslibs.nattype, pandas._libs.tslibs.timezones, pandas._libs.tslibs.tzconversion, pandas._libs.tslibs.ccalendar, pandas._libs.tslibs.fields, pandas._libs.tslibs.timedeltas, pandas._libs.tslibs.timestamps, pandas._libs.properties, pandas._libs.tslibs.offsets, pandas._libs.tslibs.parsing, pandas._libs.tslibs.conversion, pandas._libs.tslibs.period, pandas._libs.tslibs.vectorized, pandas._libs.ops_dispatch, pandas._libs.missing, pandas._libs.hashtable, pandas._libs.algos, pandas._libs.interval, pandas._libs.tslib, pandas._libs.lib, pandas._libs.hashing, pandas._libs.ops, numexpr.interpreter, pandas._libs.arrays, pandas._libs.index, pandas._libs.join, pandas._libs.sparse, pandas._libs.reduction, pandas._libs.indexing, pandas._libs.internals, pandas._libs.writers, pandas._libs.window.aggregations, pandas._libs.window.indexers, pandas._libs.reshape, pandas._libs.tslibs.strptime, pandas._libs.groupby, pandas._libs.testing, pandas._libs.parsers, pandas._libs.json, _nlopt, ezdxf.acc.mapbox_earcut, vtkmodules.vtkIOXMLParser, vtkmodules.vtkIOXML, vtkmodules.vtkRenderingCore, vtkmodules.vtkFiltersGeneral, vtkmodules.vtkFiltersExtraction, casadi._casadi, vtkmodules.vtkIOCore, vtkmodules.vtkImagingCore, vtkmodules.vtkIOImage, vtkmodules.vtkRenderingContext2D, vtkmodules.vtkRenderingFreeType, vtkmodules.vtkRenderingSceneGraph, vtkmodules.vtkRenderingVtkJS, vtkmodules.vtkIOExport, vtkmodules.vtkFiltersSources, vtkmodules.vtkInteractionWidgets, vtkmodules.vtkRenderingAnnotation, vtkmodules.vtkInteractionStyle (total: 113)
Segmentation fault      (core dumped) python -m pytest tests
```

## Core dump
```
Mar 26 16:51:59 systemd[1]: Started Process Core Dump (PID 37927/UID 0).
Mar 26 16:52:01 systemd-coredump[37928]: [🡕] Process 37898 (python) of user 1000 dumped core.
                                                
[... 219 lines cut ...]
                                                
                                                Stack trace of thread 37921:
                                                #0  0x0000780a35f2488d syscall (libc.so.6 + 0x10688d)
                                                #1  0x0000780a1e6f014c n/a (libtbb.so.12 + 0xf14c)
[... cut ...]
                                                #2  0x0000780a1e709be3 n/a (libtbb.so.12 + 0x28be3)
                                                #3  0x0000780a35ea955a n/a (libc.so.6 + 0x8b55a)
                                                #4  0x0000780a35f26a3c n/a (libc.so.6 + 0x108a3c)
                                                ELF object binary architecture: AMD x86-64

```

## Environment

OS: Arch Linux

Was CadQuery installed using Conda?: no, but via https://aur.archlinux.org/packages/python-cadquery

Using: Python interpreter

adam-urbanczyk, 2024-03-28:
Hm, does it work for you when installed via conda?

jmwright, 2024-04-04:
I cannot reproduce when installed via conda or pip with Python 3.11.