CadQuery/cadquery #1551: Export to step issue

IslamZaid, 2024-03-29:
I have built a hollow hemisphere with a small cylinder-shaped cut as shown in the figure, but when I export it as step or stl, I do not see the small cut! Can you help me on that?
[image]

and here is the saved object 

[image]

jmwright, 2024-03-29:
Please share your code.

IslamZaid, 2024-04-02:
I have figured out the issue. The problem arose from attempting to use dimensions smaller than the base unit of CadQuery's system, which is millimeters. Therefore, dimensions such as 0.001 were being ignored.

It's essential to note that CadQuery's unit system is based on millimeters. Hence, there's no need for fractions like 0.001, as they will be disregarded.