CadQuery/cadquery #1552: Sweep operation does not follow polyline

danieledapo, 2024-03-29:
I'm trying to sweep a rect over a given polyline, but I'm getting strange results. In particular the final solid does not look to be following the initial 2d path (shown in red in the image) and it does not form a single solid, but rather multiple ones with holes in them.

I've tried to use a spline rather than a polyline, but it does not change much. I've also tried some simpler paths and the code works in those cases, but I'm new to 3d modeling and it's very possible that I'm misusing some APIs :sweat_smile: 

```python
import cadquery as cq

path = [(113.49106291666668, 66.809540625),
 (113.48604399962518, 65.75564203889229),
 (113.48102283048185, 64.70174346372286),
 (113.47905083333333, 63.6478376481315),
 (113.47905083333333, 62.593927107093826),
 (113.47905083333333, 61.54001656605616),
 (113.47905083333333, 60.48610602501849),
 (113.47905083333333, 59.43219548398083),
 (113.47905083333333, 58.37828494294316),
 (113.47905083333333, 57.324374401905494),
 (113.47905083333333, 56.27046386086782),
 (113.47905083333333, 55.21655331983015),
 (113.47905083333333, 54.162642778792474),
 (113.47905083333333, 53.10873223775481),
 (113.47905083333333, 52.054821696717134),
 (113.47905083333333, 51.00091115567946),
 (113.47905083333333, 49.94700061464179),
 (113.47905083333333, 48.89309007360412),
 (113.47905083333333, 47.83917953256644),
 (113.47905083333333, 46.78526899152875),
 (113.47905083333333, 45.73135845049107),
 (113.47905083333333, 44.67744790945339),
 (113.47905083333333, 43.62353736841572),
 (113.47905083333333, 42.56962682737804),
 (113.47905083333333, 41.51571628634036),
 (113.47905083333333, 40.46180574530268),
 (113.47905083333333, 39.407895204265),
 (113.47905083333333, 38.353984663227315),
 (113.47905083333333, 37.30007412218964),
 (113.47905083333333, 36.24616358115196),
 (113.47905083333333, 35.19225304011428),
 (113.47905083333333, 34.1383424990766),
 (113.4432471548529, 33.091593949277176),
 (112.72906291666666, 32.427518541666664)]

poly = cq.Workplane("XY").polyline(path)
profile = cq.Workplane("XZ").rect(5,5)
swept = profile.sweep(poly)
```

[image]

jmwright, 2024-03-29:
The profile and the path are not aligned properly before the sweep (see
screenshot).

danieledapo, 2024-03-30:
@jmwright thanks for your quick reply! I didn't think about aligning the path and profile!

I'm trying a simpler example where the profile and the path are aligned before the sweep and it works for a simple vertical line, but as soon as I add a non vertical segment I get the same broken results. I've also tried to pass `isFrenet=True` and `makeSolid=True` but it does not change much.

```python
import cadquery as cq

path = [
# it work fine with only these two points
 (113.47905083333333, 63.6478376481315),
 (113.47905083333333, 34.1383424990766),
 
 # if I add this point it breaks
 (113.4432471548529, 33.091593949277176),
 ]

poly = cq.Workplane("XY").polyline(path)
profile = cq.Workplane("XZ").rect(5,5).translate(path[0])
swept = profile.sweep(poly)
```
[image]

jmwright, 2024-04-04:
Try this code (notice the multisection parameter):
```python
import cadquery as cq

path = [
# it work fine with only these two points
 (113.47905083333333, 63.6478376481315),
 (113.47905083333333, 34.1383424990766),
 (113.4432471548529, 33.091593949277176),
 ]

poly = cq.Workplane("XY").polyline(path)
profile = cq.Workplane("XZ").rect(5,5).translate(path[0])
swept = profile.sweep(poly, multisection=True)

show_object(swept)
```

danieledapo, 2024-04-05:
That seems to work! I completely overlooked that parameter!

I'm closing the issue then. Thank you!

closed 2024-04-05 by danieledapo