CadQuery/cadquery #1556: GMSH integration yields UContinuity error

bragostin, 2024-04-08:
Dear CQ community,

Using
 ```
cadquery @ git+https://github.com/CadQuery/cadquery.git@153ed3f667911e909ba8b93bddd297defc7cd42f
cadquery-ocp==7.7.2
gmsh==4.12.2
```
the following simple code adapted from #1346
```python
import cadquery as cq
import gmsh

FD_new = cq.Workplane("YZ").center(-2.5, -10).rect(5, 20).extrude(100)

gmsh.initialize()
gmsh.option.setNumber("General.Terminal", 1)
topods = FD_new.toOCC()
volumes = gmsh.model.occ.importShapesNativePointer(topods._address())
gmsh.model.occ.synchronize()
gmsh.option.setNumber("Mesh.Algorithm", 1)
gmsh.option.setNumber("Mesh.MeshSizeMin", 1)
gmsh.option.setNumber("Mesh.MeshSizeMax", 5)
gmsh.model.mesh.generate(3)
gmsh.write("FD_new_mesh.msh")
```
yields this error
```
Error   : OpenCASCADE exception GeomAdaptor_Surface::UContinuity
Traceback (most recent call last):
  File "/home/chbrago/Documents/Puffer_Knife/Python/Idea_Hour/issue_gmsh.py", line 11, in <module>
    volumes = gmsh.model.occ.importShapesNativePointer(topods._address())
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chbrago/cadquery/lib/python3.11/site-packages/gmsh.py", line 8627, in importShapesNativePointer
    raise Exception(logger.getLastError())
Exception: OpenCASCADE exception GeomAdaptor_Surface::UContinuity
```
any idea what is wrong here?

jmwright, 2024-04-10:
@bragostin Did you install CadQuery via pip? It would be interesting to see if using conda fixes the issue, and also if using an older version of OCP helps (to test for a regression in OpenCASCADE/OCP. If neither of those help, maybe you can check with the gmsh community to see if there is any extra object debugging that can be done on that side?

bragostin, 2024-04-10:
@jmwright Yes it was a `pip`install.
Reverting to `cadquery-ocp==7.7.1` does not solve the issue.
Export in a STEP file and importing via `gmsh.merge` function works.
I will try with a conda install next.

bragostin, 2024-04-11:
@jmwright  with a fresh conda install `conda create -n cadquery -c conda-forge -c cadquery cadquery=master cq-editor=master` the issue disappears. 
So the problem comes from the pip install it seems.

adam-urbanczyk, 2024-04-11:
Then it is likely an ABI compatibility issue. conda-forge avoids this by having a well defined toolchain. That is not the case for pip, everyone can choose a different compiler/flags.

bragostin, 2024-04-11:
@adam-urbanczyk thank you! Anything that can be done beyond using conda? Like raising the issue with the `gmsh` team?

adam-urbanczyk, 2024-04-11:
You could try compiling all packages yourself. I don't think this is a gmsh or ocp packaging bug.