CadQuery/cadquery #1557: An error is reported when the code is run

binfengjiao, 2024-04-09:
code：
# Python
import cadquery as cq
from lq.topologies.schwartz import schwartz_p_heterogeneous_lattice
cq.Workplane.schwartz_p_heterogeneous_lattice = schwartz_p_heterogeneous_lattice

# BEGIN USER INPUT
unit_cell_size = 4
Nx = 10
Ny = 10
Nz = 10
min_thickness = 0.1
max_thickness = 7
# END USER INPUT

schwartz = schwartz_p_heterogeneous_lattice(unit_cell_size, min_thickness, max_thickness,
                                      Nx, Ny, Nz
                                      )

Error：
ValueError: STEP File could not be loaded

jmwright, 2024-04-10:
@binfengjiao There is a lot of code that we cannot see, so it makes it impossible to help. Can you post a minimal example that reproduces the issue?

binfengjiao, 2024-04-11:
Hello, thank you very much for your help, can you connect my computer with a remote software called todesk and view it directly on my computer? I suspect it's a problem with the environment configuration. The account number is 438 467 087, and the password is ***@***.***

jmwright, 2024-04-11:
@binfengjiao I have deleted your last comment sent via email because it contained login information. We are an open source project of volunteers. We do not provide the sort of remote support that you asked for.