CadQuery/cadquery #1559: How to solve the error

binfengjiao, 2024-04-11:
How to solve the error in the picture? The specific content of the reference function is as follows:
def bcc_heterogeneous_lattice(unit_cell_size,
                              min_strut_diameter,
                              max_strut_diameter,
                              min_node_diameter,
                              max_node_diameter,
                              Nx, Ny, Nz,
                              topology='bcc',
                              rule='linear',
                              position=(0, 0, 0),
                              rotation=(0, 0, 0)):
    if topology not in ['bcc', 'bccz', 'sbcc', 'sbccz']:
        raise TypeError(f'The type \'{topology}\' does not exist!')
    min_strut_radius = min_strut_diameter / 2.0
    max_strut_radius = max_strut_diameter / 2.0
    if rule == 'linear':
        strut_radii = np.linspace(min_strut_radius,
                                  max_strut_radius,
                                  Nz)
        node_diameters = np.linspace(min_node_diameter,
                                     max_node_diameter,
                                     Nz)
    if rule == 'sin':
        average = lambda num1, num2: (num1 + num2) / 2
        strut_radii = np.sin(
            np.linspace(min_strut_radius, max_strut_radius, Nz) * 12) + 2 * average(min_strut_radius, max_strut_radius)
        node_diameters = np.sin(
            np.linspace(min_node_diameter, max_node_diameter, Nz) * 12) + 2 * average(min_node_diameter,
                                                                                      max_node_diameter)
    if rule == 'parabola':
        x = np.linspace(0, 1, num=Nz)
        frep = lambda d_min, d_max: -4 * d_max * (x - 0.5) * (x - 0.5) + d_max + d_min
        strut_radii = frep(min_strut_radius, max_strut_radius)
        print(strut_radii)
        node_diameters = frep(min_node_diameter, max_node_diameter)
    UC_pnts = []
    for i in range(Nx):
        for j in range(Ny):
            for k in range(Nz):
                UC_pnts.append((i * unit_cell_size, j * unit_cell_size, k * unit_cell_size))
    print("Datapoints generated")
    result = cq.Workplane().tag('base').transformed(
        offset=cq.Vector(position[0], position[1], position[2]),
        rotate=cq.Vector(rotation[0], rotation[1], rotation[2]))
    result = result.pushPoints(UC_pnts)
    unit_cell_params = []
    for i in range(Nx * Ny):
        for j in range(Nz):
            unit_cell_params.append({"unit_cell_size": unit_cell_size,
                                     "strut_radius": strut_radii[j],
                                     "node_diameter": node_diameters[j],
                                     "type": topology})
    result = result.eachpointAdaptive(unit_cell,
                                      callback_extra_args=unit_cell_params,
                                      useLocalCoords=True)
    print("The lattice is generated")
    return result

[image]

[image]
[image]

huskier, 2024-04-15:
I’m very curious what the no_MRE stands for? I guess it's **no Minimal Reproducible Example** (no_MRE). Right?

adam-urbanczyk, 2024-04-15:
Indeed, you are right.

huskier, 2024-04-15:
Thanks. @adam-urbanczyk 

@binfengjiao You may firstly put the code inside the CODE BLOCK. You'd better to make a MRE to pinpoint the issue.
```
[the code]
```