CadQuery/cadquery #1562: Generate a closed solid from a 2d spline doing Face -> Shell -> Solid

nfoppiani-pf, 2024-04-15:
I am trying to generate a torus-like solid from points using 2d splines, and I am having the following issue when moving from a shell to a solid.
```
cad_solid = cq.Solid.makeSolid(cad_shell)
  File "/home/nico/.cache/pants/named_caches/pex_root/venvs/s/08cb250b/venv/lib/python3.10/site-packages/cadquery/occ_impl/shapes.py", line 3068, in makeSolid
    return cls(ShapeFix_Solid().SolidFromShell(shell.wrapped))
TypeError: SolidFromShell(): incompatible function arguments. The following argument types are supported:
    1. (self: OCP.ShapeFix.ShapeFix_Solid, shell: OCP.TopoDS.TopoDS_Shell) -> OCP.TopoDS.TopoDS_Solid

Invoked with: <OCP.ShapeFix.ShapeFix_Solid object at 0x7dc1d58a73f0>, <OCP.TopoDS.TopoDS_Face object at 0x7dc20a6d8a30>
```

Can you help me understand why, despite the object is indeed a cq.Shell, it seems it's actually a TopoDS_Face for OCP?

Thanks!

## To Reproduce

```python

import cadquery as cq
import numpy as np

if __name__ == "__main__":

    # define a torus
    major_radius = 1000.0
    minor_radius = 100.0

    phis = np.linspace(0, 2 * np.pi, 101, endpoint=False)[:, np.newaxis]
    thetas = np.linspace(0, 2 * np.pi, 100, endpoint=False)[np.newaxis, :]

    x = (major_radius + minor_radius * np.cos(thetas)) * np.cos(phis)
    y = (major_radius + minor_radius * np.cos(thetas)) * np.sin(phis)
    z = minor_radius * np.sin(thetas) * np.ones_like(phis)

    list_of_list_of_vectors: list[list[cq.Vector]] = []
    for i_phi in range(len(phis)):
        aux: list[cq.Vector] = []
        for i_theta in range(len(thetas)):
            aux.append(
                cq.Vector(x[i_phi, i_theta], y[i_phi, i_theta], z[i_phi, i_theta])
            )
        aux.append(aux[0])
        list_of_list_of_vectors.append(aux)
    list_of_list_of_vectors.append(list_of_list_of_vectors[0])

    cad_surface = cq.Face.makeSplineApprox(list_of_list_of_vectors)
    cad_shell = cq.Shell.makeShell([cad_surface])
    cad_solid = cq.Solid.makeSolid(cad_shell)
```

## Full Backtrace
```
Traceback (most recent call last):
  File "/tmp/pants-sandbox-pi26mO/./.cache/pex_root/venvs/b11122f7e072e8bdad33fdf5ced2484f67ee7b26/c762b2e34782c98c4694293366adaa58e858ecad/pex", line 274, in <module>
    runpy.run_module(module_name, run_name="__main__", alter_sys=True)
  File "/home/nico/.pyenv/versions/3.10.12/lib/python3.10/runpy.py", line 224, in run_module
    return _run_module_code(code, init_globals, run_name, mod_spec)
  File "/home/nico/.pyenv/versions/3.10.12/lib/python3.10/runpy.py", line 96, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "/home/nico/.pyenv/versions/3.10.12/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/tmp/pants-sandbox-pi26mO/./experimental/nfoppiani/parametric_cad/surface_2d_spline_cadquery_github_issue.py", line 44, in <module>
    cad_solid = cq.Solid.makeSolid(cad_shell)
  File "/home/nico/.cache/pants/named_caches/pex_root/venvs/s/08cb250b/venv/lib/python3.10/site-packages/cadquery/occ_impl/shapes.py", line 3068, in makeSolid
    return cls(ShapeFix_Solid().SolidFromShell(shell.wrapped))
TypeError: SolidFromShell(): incompatible function arguments. The following argument types are supported:
    1. (self: OCP.ShapeFix.ShapeFix_Solid, shell: OCP.TopoDS.TopoDS_Shell) -> OCP.TopoDS.TopoDS_Solid

Invoked with: <OCP.ShapeFix.ShapeFix_Solid object at 0x7dc1d58a73f0>, <OCP.TopoDS.TopoDS_Face object at 0x7dc20a6d8a30>
```

adam-urbanczyk, 2024-04-15:
Currently `makeSplineApprox` does not work for periodic surfaces.

nfoppiani-pf, 2024-04-15:
Understood. However, the shell turns out out to be correct if exported to .step or .stl. Would it be possible to find a workaround for me to generate a solid in the meantime?

adam-urbanczyk, 2024-04-15:
The `cad_surface` has 1 edge and 0 area. Does not sound correct to me. I don't understand your use case, but it does not seem to be supported by CQ. You'll need to use the OCP APIs directly.

nfoppiani-pf, 2024-04-15:
My use case is to generate closed toroidal-like geometries by creating 2d splines through the points. I don't fully understand why the area is zero. If you export the shell as a step or stl file, you can see it correctly reproduces the toroidal geometry.