CadQuery/cadquery #1564: Importing Cadquery causes error exit codes

JacobAkimbo, 2024-04-18:
## To Reproduce
```python
import cadquery

pass
```
Doing this causes the system to finish with exit code (0xC0000005) or exit code (0xc0000374). Version 2.3.1 and 2.4.0 respectively.

## Environment

OS: Windows
CadQuery was installed with pip in an empty venv running python 3.11.9

huskier, 2024-04-19:
On my OS, there is no problem.
The OS:
> Windows 11 professional
> 23H2
> 22631.3447

The Python distribution:
> Anaconda3-2023.09-0-Windows-x86_64.exe
> Python 3.11.5

The CadQuery version: 
> Python 3.11.5 | packaged by Anaconda, Inc. | (
> >>> import cadquery
> >>> print(cadquery.__version__)
> 2.4.0

To reproduce, a test.py file is created, and its content is as following:

> import cadquery
> pass

huskier, 2024-04-20:
On my side, there is no problem.
The OS:
> Windows 11 professional
> 23H2
> 22631.3447

The Python distribution:
> Anaconda3-2023.09-0-Windows-x86_64.exe
> Python 3.11.5

The Python venv is:
> (cqvenv) (base) D:\>python
> Python 3.11.5 | packaged by Anaconda, Inc. | (main, Sep 11 2023, 13:26:23) [MSC v.1916 64 bit (AMD64)] on win32
> >>>

The CadQuery version: 
(cqvenv) (base) D:\>python
Python 3.11.5 | packaged by Anaconda, Inc. | (main, Sep 11 2023, 13:26:23) [MSC v.1916 64 bit (AMD64)] on win32
> import cadquery
> print(cadquery.__version__)
2.4.0
>>>

To reproduce, a test.py file is created, and its content is as following:

> import cadquery
> pass

jmwright, 2024-04-21:
@jdegenstein Could this be related to the cadquery-ocp issue that you pinged me about? 

https://github.com/CadQuery/ocp-build-system/pull/26

akaszynski, 2024-09-12:
While not directly related to the above issue, I ran into a similar problem when using pytest on `cadquery` 5429f618c6723b47f07ba5394a912e22b0dd5cff on Windows 10 on Python 3.10 with `pytest`.

```
platform win32 -- Python 3.10.9, pytest-8.2.2, pluggy-1.5.0
rootdir: C:\Users\Alex\<redacted>
configfile: pyproject.toml
plugins: repeat-0.9.3, rerunfailures-14.0
collected 13 items

tests\test_cad.py .............                                                        [100%]

==================================== 13 passed in 29.51s ====================================
Windows fatal exception: access violation

Current thread 0x00002504 (most recent call first):
  <no Python frame>
```

Note that a similar script without `pytest` would work just fine.

Solution was to disable pytest's `faulthandler` with:

```
pytest -p no:faulthandler
```

Hopefully someone else finds this helpful. Note that this issue does not occur on `cadquery==v2.4.0`.

adam-urbanczyk, 2024-09-12:
Thanks @akaszynski , could you paste what is in your env (e.g. `conda list` or similar)?

gdahlberg55, 2026-09-03:
Ran into this today on Windows 11 (Python 3.11) with cadquery 2.8.0 and 2.6.1 installed from PyPI: every process that imports cadquery finishes its work normally, then dies during interpreter shutdown with `0xC0000374` (STATUS_HEAP_CORRUPTION) or `0xC0000005`. WER minidumps from dozens of these all show the same faulting stack inside `_nlopt.pyd` teardown → ucrtbase → ntdll heap functions.

Isolated it to the nlopt × casadi pair — nothing else is needed:

| command | casadi 3.8.0 + nlopt 2.11.0 | casadi 3.7.2 + nlopt 2.11.0 |
|---|---|---|
| `python -c "import nlopt, casadi"` | exit 0xC0000374 | exit 0 |
| `python -c "import cadquery"` | exit 0xC0000374 | exit 0 |
| `import casadi` alone, `import nlopt` alone, `import nlopt, OCP.gp` | exit 0 | exit 0 |

So on this machine the crash-at-exit appears whenever casadi >= 3.8.0 and nlopt 2.11.0 are loaded in the same process (importing cadquery always loads both); pinning `casadi<3.8` makes everything exit clean, verified in two separate venvs. cadquery itself works fine either way — the corruption is purely in the exit path. Since cadquery's metadata leaves `casadi` unbounded, a fresh `pip install cadquery` today pulls the crashing combination by default.

Beyond the WER-dump noise, the practical damage is that anything supervising a cadquery script (CI, task runners) sees a crash exit code after a fully successful run.

Happy to share the minidump stack details if useful.

adam-urbanczyk, 2026-09-04:
@gdahlberg55 could you share the dump for completeness?