CadQuery/cadquery #1567: CQ-Editor doesn't update when using external editor (gvim or notepad) on windows 11

gitterman, 2024-04-21:
I prefer to use gvim instead of the built-in editor of CQ-Editor.
After my initial installation of CQ-Editor I could save a model.py file and automatically saw the updated preview in CQ-Editor.
When I now save a model.py file, reload and preview doesn't work.
Sometimes I get a FileNotFoundError, but when I then open the file from the "Open" icon in CQ-Editor, it reloads and previews the result properly.
I tried with some changes in the auto-reload settings but no success.
With other apps, e.g. the Arduino IDE or Thonny, it is no problem to use an external editor.
They instantly update if I do a save in an external editor.
How can I get CQ-Editor to work with an external editor too?

adam-urbanczyk, 2024-04-21:
I don't have windows 11, but on Win 10 and with vim all works as intended.

jmwright, 2024-04-21:
@gitterman You could try adjusting the auto-reload delay. 

https://github.com/CadQuery/CQ-editor?tab=readme-ov-file#using-an-external-code-editor

Some editors delete and recreate the file when it is saved, and maybe the timing is a little bit off.

gitterman, 2024-04-21:
@jimwright: Thanks a lot for the hint, the error is gone.
I had decreased Autoreload delay because I assumed from the name, that a shorter delay would increase the responsiveness.
But it obviously is the other way round. A higher delay assures that an externally edited file gets more properly recognized when changed.
_4. If CQ-editor is not catching the saves from your external editor, increasing Autoreload delay in the right pane may help. This issue has been reported when using vim or emacs._

closed 2024-04-21 by gitterman