CadQuery/cadquery #1572: About the usage of CQModel?

huskier, 2024-04-26:
We want to create a 3D model from a string liking the following code. 
```
import cadquery as cq
import textwrap
from cadquery.vis import show
from cadquery import cqgi

TESTSCRIPT = textwrap.dedent(
    """
        result_in_TESTSCRIPT =  cq.Workplane().box(50, 50, 5)
        cq.exporters.export(result_in_TESTSCRIPT, "result_in_TESTSCRIPT.step", exportType = "STEP")        
        show(result_in_TESTSCRIPT)
    """
)

model = cqgi.CQModel(TESTSCRIPT)
result = model.build()

print(result.success)
print(type(result))
print(len(result.results))
```

After "model.build()", we've gotten the exported "result_in_TESTSCRIPT.step" file; however, the "show(result_in_TESTSCRIPT)" function does not show the 3D model, meanwhile, result.success is printed "False".

When we comment the "show(result_in_TESTSCRIPT)" function, the result.success is printed "True".

Another question: After "model.build()", could we obtain the 3D model of "result_in_TESTSCRIPT" for subsequent operations?

The text in the source code says this is possible.
```
class CQModel(object):
    """
    Represents a Cadquery Script.

    After construction, the metadata property contains
    a ScriptMetaData object, which describes the model in more detail,
    and can be used to retrieve the parameters defined by the model.

    the build method can be used to generate a 3d model
    """
```

jmwright, 2024-04-26:
You are not importing `show` inside of your code text block. This modification fixes the error and prints the exception when there is one.
```python
import cadquery as cq
import textwrap
from cadquery.vis import show
from cadquery import cqgi

TESTSCRIPT = textwrap.dedent(
    """
        from cadquery.vis import show
        result_in_TESTSCRIPT =  cq.Workplane().box(50, 50, 5)
        cq.exporters.export(result_in_TESTSCRIPT, "result_in_TESTSCRIPT.step", exportType = "STEP")        
        show(result_in_TESTSCRIPT)
    """
)

model = cqgi.CQModel(TESTSCRIPT)
result = model.build()

if not result.success:
    print(result.exception)

# print(result.success)
print(type(result))
print(len(result.results))
```

huskier, 2024-04-26:
I mis-think that the third line code "from cadquery.vis import show" do the job. Yes, the first issue is resolved. Thank you, @jmwright.

After executing "model.build()", could we obtain the model of result_in_TESTSCRIPT in the main program (not the TESTSCRIPT script) for the subsequent operations?

jmwright, 2024-04-26:
[This code](https://github.com/jmwright/cadquery-freecad-module/blob/master/CQGui/Command.py#L204) is older and unmaintained currently, but is still relevant and gives an example of extracting CQ objects from the CQGI build result object, which would allow you to use them later.

huskier, 2024-04-26:
Thank you very much for you help. @jmwright 

My issues are resolved, and the final code is as following.
```
import cadquery as cq
import textwrap
from cadquery.vis import show
from cadquery import cqgi

TESTSCRIPT = textwrap.dedent(
    """
        result_in_TESTSCRIPT =  cq.Workplane().box(50, 50, 5)
        cq.exporters.export(result_in_TESTSCRIPT, "result_in_TESTSCRIPT.step", exportType = "STEP")        
        show_object(result_in_TESTSCRIPT)
    """
)

model = cqgi.CQModel(TESTSCRIPT)
result = model.build()

if not result.success:
    print(result.exception)
else:
    print("success......")
    for res in result.results:
        print(type(res.shape))
        show(res.shape)
```

closed 2024-04-26 by huskier