CadQuery/cadquery #1576: Finding coplanar and concentric surfaces

mxmws, 2024-04-29:
I imported a 3D part from a step file. Addionally I have parameters for the following surfaces:
1. a plane defined by a directional vector and an offset
2. a cylinder surface defined by a positional vector, a directional vector and a radius

I would like to find all faces in my part that are either coplanar to my plane or concentric and equal in radius to my cylinder surface.

My idea was to iterate over all faces and compare their parameters with the plane/cylinder I have.
Is there a better way to do this and if no, how do I iterate over the faces of my part?

bragostin, 2024-05-02:
Personally I use `BoxSelector`, like this
```python
faces = cq_object.faces(cq.selectors.BoxSelector(corner0, corner1, boundingbox=True))
```
to select objects having the same, or contained in,  bounding boxes.

mxmws, 2024-05-02:
> Personally I use `BoxSelector`, like this
> 
> ```python
> faces = cq_object.faces(cq.selectors.BoxSelector(corner0, corner1, boundingbox=True))
> ```
> 
> to select objects having the same, or contained in, bounding boxes.

Thank you, this is helpful! Do you have any idea how I could do the same for cylinders?

bragostin, 2024-05-02:
Maybe comparing areas and centers with
```python
center = face.val().Center().toTuple()
area = face.val().Area()
```

mxmws, 2024-05-06:
> Personally I use `BoxSelector`, like this
> 
> ```python
> faces = cq_object.faces(cq.selectors.BoxSelector(corner0, corner1, boundingbox=True))
> ```
> 
> to select objects having the same, or contained in, bounding boxes.

Thank you again, this works. But how can I select the remaining surfaces? "not" doesn't work. Basically I want to show all surfaces but color the selected ones.

bragostin, 2024-05-10:
You could get a list of all surfaces with `faces = cq_object.faces().vals()` and then remove from this list the selected ones. That would leave you with a list of not selected surfaces.