CadQuery/cadquery #1584: Simple 'cut' of two a Box and a Sphere creates unusable object

fleiner, 2024-05-08:
This very simple script creates a box and cuts out part of it with a sphere. It confuses cq-editor (instead of a cut one gets a hole to look into the cube), and creates a completely unusable stl file with over 1000 open edges,  (the equivalent  translate() call  instead of transformGeometry() is slightly better with only 4  open edges). Is there any workaround? Thanks

```python
import cadquery as cq

res=cq.Solid.makeBox(10,10,10)
sphere=cq.Solid.makeSphere(4,angleDegrees1=-90)

m=[[1,0,0,10],[0,1,0,0],[0,0,1,5],[0,0,0,1]]
res=res.cut(sphere.transformGeometry(cq.Matrix(m)))
#res=res.cut(sphere.translate([10,0,5]))

ass = cq.Assembly()
ass.add(res, color=cq.Color("red"))

cq.exporters.export(cq.Workplane("XY").add(res), 'mesh.stl')
show_object(ass)
```
[image]

adam-urbanczyk, 2024-05-08:
The workaround would be to perturb the position of the sphere: `res=res.cut(sphere.translate([10,1e-3,5]))`

fleiner, 2024-05-08:
That does solve the cq-editor problem, but it does not solve the stl code. For the latter I have to perturb all three axis.

jmwright, 2024-05-13:
The following code using the fluent API works fine for me (the STL imports and slices without any errors).
```python
import cadquery as cq

# Box
box = cq.Workplane().box(10, 10, 10)

# Sphere
sp = cq.Workplane().sphere(4)
sp = sp.translate((5, 5, 0))

# Resulting object
res = box.cut(sp)

show_object(res)

cq.exporters.export(res, "/home/jwright/Downloads/cut_obj.stl")
```