CadQuery/cadquery #1585: Documentation: Assembly tutorial constraint solver result regression

Roger-random, 2024-05-12:
As a CadQuery beginner I've been hanging out in documentations. I noticed a regression in the Assemblies documentation page introduced sometime in the past week(-ish) timeframe. Every constraint solver result shown in the 3D viewer seems to have gone awry. GitHub says the documentation page source (https://github.com/CadQuery/cadquery/blob/master/doc/assy.rst) has not been touched in four months. I don't know where to look next.

URL: https://cadquery.readthedocs.io/en/latest/assy.html

### Expected: layout to match the static screenshot (repeated here for easy reference)

[image]

### Actual: layout now looks like this.

[image]

### Environment

Verified on following browsers:
* Firefox 125.0.3
* Chrome 124.0.6367.202
* Edge 124.0.2478.80

OS: Windows 11

adam-urbanczyk, 2024-05-12:
Thanks, looks like the free function PR did break something. AFAICT rendering assemblies locally works fine. I guess some kind of weird interaction between the cq directive and the new `cq.Location` constructors. I will look into it.

lorenzncode, 2024-05-12:
I confirm local render with cq-editor or `cadquery.vis` works fine here.  I can reproduce the issue with a local sphinx doc build.

lorenzncode, 2024-05-12:
`toTuple` was changed to return degrees.
https://github.com/CadQuery/cadquery/blob/995d1393bd3cba2061da7e7d2bea8e4fdbdd19c9/cadquery/occ_impl/geom.py#L1059

`toTuple` is called here in `toJSON` used in the cq directive:
https://github.com/CadQuery/cadquery/blob/995d1393bd3cba2061da7e7d2bea8e4fdbdd19c9/cadquery/occ_impl/assembly.py#L338

adam-urbanczyk, 2024-05-12:
Yup, just opened a PR

PR #1587 (2024-05-12): Fix units in toJSON
Fixes #1585
closed 2024-05-12 by adam-urbanczyk