CadQuery/cadquery #1586: seems parametricCuve（）bug

Lofusy, 2024-05-12:
Hi, 
i think i met with bug .I want to sweep along 3D path defined by parametricCurve(), It's Basically double helix—— wire make helix revolve a normal helix.
However，when I increase t, the curve goes wrong, and I also try to increase N, It,s dosen' work too.
And I am new to cq and code thing, I need to find way to solve this prblem.

q = -1 
p = -1 
origine_angle = 0
def double_helix(r, R, B, n, t):
    x = r * (cos(origine_angle + p * q * t * n) * cos(t) - q * sin(origine_angle + p * q * t * n) * cos(B) * sin(t)) + R * cos(t)
    y = q * r * (cos(origine_angle + p * q * t * n) * sin(t) + q * sin(origine_angle + p * q * t * n) * cos(B) * cos(t)) + R * sin(t)
    z = -1*q*r*sin(origine_angle+p*q*t*n)*sin(B)+t*R/tan(B)
    return x, y, z

x, y, z = double_helix(R_22, R_21, a_21, n_22, 0)
path = (
            cq.Workplane("XY")
            .parametricCurve(lambda t: double_helix(R_22, R_21, a_21, n_22,t*2*pi),start= 0,stop= 3/4
                             )
            )

show_object(path)

when t=3/4, the curve is good, but t=1, the curve isn't the correct result.
[image]
[image]
[image]

I think that might be bug, and how to solve this.
And forgive my English,if i have ever make uncomefortabl words.

Regrads

Lofuys, 2024-05-12:
sorry， I mean t is stop

Lofuys, 2024-05-12:
[image]
and along the correct curve the result also strange

lorenzncode, 2024-05-12:
The code example is corrupted as shown in the comment.

For example ```-1qrsin```? here:
```py
z = -1qrsin(origine_angle+pqtn)sin(B)+tR/tan(B)
```

Could you repost with code formatting? See the github docs here: 
https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code

Please also include a minimal reproducible example that defines all variables used in the code.  Missing variables include ```R_22, R_21, a_21, n_22```.

Lofuys, 2024-05-13:
sorry, I am alos new to github.Its kind of tricky for me at least now.
anyway,the code is below:

import cadquery as cq
from math import sin, cos, pi, tan

def parameter_n(r_1,r_2,a,B):
    return(
        r_1 *  tan(a) / (r_2 * sin(B) )
        )

#diamater of wires , helix radius and lay angle.

r_B = 0.135

R_21 = 1.100
a_21 = 0.16309787547

R_22 = 0.27
a_22 = 0.23454060987
n_22 = parameter_n(R_21,R_22,a_22,a_21)

oreigine_angle = 0 
q=1
def single_helix(r,t,a):
    return(
        r * cos(q * t + oreigine_angle),
        r * sin(q * t + oreigine_angle),
        r * t/tan(a),
         )
x,y,z= single_helix(R_21,0,a_21)
    
path= (
        cq.Workplane("XY")
        .parametricCurve(lambda t: single_helix(R_21,t*2*pi,a_21)
                         )
        )
ISCW = (
    cq.Workplane("XY")
    .center(x,y)
    .circle(r_A)
    .sweep(path,isFrenet=True)
    )
show_object(ISCW)
        
q = -1
p = -1
origine_angle = 0
def double_helix(r, R, B, n, t):
    x = r * (cos(origine_angle + p * q * t * n) * cos(t) - q * sin(origine_angle + p * q * t * n) * cos(B) * sin(t)) + R * cos(t)
    y = q * r * (cos(origine_angle + p * q * t * n) * sin(t) + q * sin(origine_angle + p * q * t * n) * cos(B) * cos(t)) + R * sin(t)
    z = -1 * q * r * sin(origine_angle + i * 2 * pi / n + p * q * t * n) * sin(B) + t * R / tan(B)
    return x, y, z

x, y, z = double_helix(R_22, R_21, a_21, n_22, 0)

path = (
        cq.Workplane("XY")
        .parametricCurve(lambda t: double_helix(R_22, R_21, a_21, n_22,t*2*pi),start= 0,stop= 3/4
                         )
        )
show_object(path)

adam-urbanczyk, 2024-05-14:
I get `r_A` is not defined.

Lofuys, 2024-05-14:
Sorry for my mistake, it's actually  r_B. I hvae already fixed it, seeing the code above.

adam-urbanczyk, 2024-05-14:
Another error:
```NameError: name 'i' is not defined```

Lofuys, 2024-05-15:
I have fixed and test the code again, it should work now.
Apology for the errors and thaks for your patience!

adam-urbanczyk, 2024-05-16:
It seems that you need to turn off smoothing:
```python
path1 = (
    cq.Workplane("XY")
    .parametricCurve(
        lambda t: double_helix(R_22, R_21, a_21, n_22,t*2*pi),start= 0,stop= 1, smoothing=None
    )
)
```
The other thing is related to rendering tolerance (`Edit>Preferences>3D Viewer>Deviation` in CQ-editor). If I set tighter tol, I see no artifacts (though it takes longer to display the object).

closed 2024-05-28 by adam-urbanczyk