CadQuery/cadquery #1599: Working with TDocStd_Document: Empty document when trying to use TDocStd_Application::Open wrapped by OCP

GabrielJMS, 2024-05-29:
I'm trying to read/open an XDE/OCAF XML file.  I've tried already different ways of doing it. First, I've tried to read from the disk a .xml file, using:
`TDocStd_Application::Open`

 and in my last attempt I've tried to read from an io.BytesIO() that comes from my C++ code also using TDocStd_Application::Open.
 
 Bellow my code:
 
 
```
def read_ocaf_file():
    output_stream = io.BytesIO()

    # Save the document contents to the provided output stream
    success = open_ocaf_file(output_stream) # C++ binded fuction that returns a stream containing the data of the TDocStd_Document

    if success:
        # prepare a doc
        app = XCAFApp_Application.GetApplication_s()

        doc = TDocStd_Document(TCollection_ExtendedString("XmlXCAF"))
        app.InitDocument(doc)
        #app = XCAFApp_Application.GetApplication_s()
        tool = XCAFDoc_DocumentTool.ShapeTool_s(doc.Main())
        tool.SetAutoNaming_s(False)
        ctool = XCAFDoc_DocumentTool.ColorTool_s(doc.Main())
        store = XmlXCAFDrivers_DocumentStorageDriver(
            TCollection_ExtendedString("Copyright: Open Cascade, 2001-2002")
        )
        ret = XmlXCAFDrivers_DocumentRetrievalDriver()

        app.DefineFormat(
            TCollection_AsciiString("XmlXCAF"),
            TCollection_AsciiString("Xml XCAF Document"),
            TCollection_AsciiString("xml"),
            ret,
            store,
        )
        #doc.UnsetIsReadOnly()
        status = app.Open(io.BytesIO(output_stream.getvalue()), doc)
        assert status == PCDM_StoreStatus.PCDM_SS_OK
        
    else:
        print("Failed to save the document.")

    labels = TDF_LabelSequence()
    tool.GetFreeShapes(labels)
    shapes = []
    for i in range(1, labels.Length() + 1):
        label = labels.Value(i)
        shape = self.shape_tool.GetShape_s(label)
        shapes.append(shape)

    app.Close(doc)
```

Although, at the end the app.Open returns PCDM_StoreStatus.PCDM_SS_OK when I try to get data from the doc is like it's empty, If I try to do for example: app.Close(doc), I get this following error:
```
    app.Close(doc)
OCP.Standard.Standard_Failure: cannot close a document that has not been opened
```

Does anyone have experience on working with OCAF/XDE files and has experienced similar difficulties with TDocStd_Document?

Any insights or examples on this matter would be highly appreciated!

Thank you very much for the attention

GabrielJMS, 2024-05-29:
Here the .xml file that I'm trying to read: [test.zip](https://github.com/CadQuery/cadquery/files/15485062/test.zip)

alfyorov, 2024-07-13:
I have the same problem working with the BinXCAF format.
```
from OCP.BinXCAFDrivers import BinXCAFDrivers
from OCP.Message import Message_ProgressRange
from OCP.TCollection import TCollection_ExtendedString
from OCP.TDocStd import TDocStd_Application, TDocStd_Document

app = TDocStd_Application()
BinXCAFDrivers().DefineFormat_s(app)

doc = TDocStd_Document(TCollection_ExtendedString("BinXCAF"))

app.NewDocument(TCollection_ExtendedString("BinXCAF"), doc)

open_status = app.Open(TCollection_ExtendedString("test.xbf"), doc, Message_ProgressRange())
print("open_status", open_status)

store_status = app.SaveAs(doc, TCollection_ExtendedString("output.xbf"), Message_ProgressRange())
print("store_status", store_status)
```

It logs:
```
open_status PCDM_ReaderStatus.PCDM_RS_OK
store_status PCDM_StoreStatus.PCDM_SS_No_Obj
```

Here is the test document: [test.zip](https://github.com/user-attachments/files/16201988/test.zip)