CadQuery/cadquery #1605: How to calculate moment of inertia?

lenianiva, 2024-06-14:
If I have an assembly where different parts have different colours, and each colours corresponds to a density, is there a way to calculate the MoI about a particular axis?

bragostin, 2024-06-15:
I guess your question regarding CadQuery is how to get the volume of an object so that you can calculate the moment of inertia? That would be `cq_object.val().Volume()`

lenianiva, 2024-06-15:
> I guess your question regarding CadQuery is how to get the volume of an object so that you can calculate the moment of inertia? That would be `cq_object.val().Volume()`

No I need to calculate the MoI of an assembly of objects with different densities. If the object is homogeneous I can just call `matrixOfInertia()`

adam-urbanczyk, 2024-06-28:
You'll need to implement this yourself

closed 2024-07-14 by lenianiva