CadQuery/cadquery #1609: Sweep or extrude flips the object when plane is 45 degrees

Jopie01, 2024-06-20:
I walk into the situation where I do a sweep or extrude from a plane which has an angle of 45 degrees. At that moment the object is flips. Use the code below for reproduction.

```python
import cadquery as cq
from math import radians, cos, sin

length = 1000 # mm
angle = 48 # degrees <--- play with this use 45 and 45.1 to see the flip
angle_radian = radians(angle)

path = (cq.Workplane("YZ")
    .lineTo(length * cos(angle_radian), length * sin(angle_radian))
)

plane = cq.Plane(origin=path.val().startPoint(), normal=path.val().tangentAt(1))

plate = (cq.Workplane(plane)
    .lineTo(0, 155)
    .lineTo(5, 155)
    .lineTo(5, 5)
    .lineTo(140, 5)
    .lineTo(140, 0)
    .close()
)
#res = plate.sweep(path)
res = plate.extrude(length)
show_object(res)
```

I draw a line with an angle and put a plane at the startpoint which is normal to that line. Then I use that plane to draw a L-shaped sketch and sweep it or extrude it. In both cases the result its flipped.

lorenzncode, 2024-06-21:
`xDir` is optional when creating the plane.  It is computed automatically if not specified.

https://github.com/CadQuery/cadquery/blob/33f72d2e5747d7996c5d789c55e2febe6b5d2167/cadquery/occ_impl/geom.py#L565-L567

Try with explicit xDir:

```py
plane = cq.Plane(
    origin=path.val().startPoint(),
    xDir=(1, 0, 0),
    normal=path.val().tangentAt(0),
)
```

Jopie01, 2024-06-21:
Indeed, using explicitly the `xDir`, everything stays where it should be.

Thanks again.

closed 2024-06-21 by Jopie01