CadQuery/cadquery #1616: had to downgrade numpy from 2.0.0 to 1.26.4 to make cadquery work

MotoCyti, 2024-06-28:
new install of cadquery on windows 10.  Cadquery version 2.4.0
using python 3.12.4
test of install  with "import cadquery" produced an error: AttributeError: module 'numpy' has no attribute 'bool8'.  Did you mean: 'bool'
google search returned that latest numpy no longer has bool8.
So I downgraded numpy from 2.0.0 to 1.26.4
It solve the problem.  basic cadquery test passed.

closed 2024-06-28 by adam-urbanczyk

drayde, 2024-07-10:
#1617 did not fix the issue for me, there is still a problem with numpy2:

Using the current dev version of cadquery doesn't do the trick, there's still an issue:

> 
> /opt/hostedtoolcache/Python/3.11.9/x64/lib/python3.11/site-packages/cadquery/occ_impl/sketch_solver.py:10: in <module>
>     import nlopt
> /opt/hostedtoolcache/Python/3.11.9/x64/lib/python3.11/site-packages/nlopt/__init__.py:1: in <module>
>     from .nlopt import *
> /opt/hostedtoolcache/Python/3.11.9/x64/lib/python3.11/site-packages/nlopt/nlopt.py:20: in <module>
>     from . import _nlopt
> E   ImportError: numpy.core.multiarray failed to import
> ------------------------------- Captured stderr --------------------------------
> 
> A module that was compiled using NumPy 1.x cannot be run in
> NumPy 2.0.0 as it may crash. To support both 1.x and 2.x
> versions of NumPy, modules must be compiled with NumPy 2.0.
> Some module may need to rebuild instead e.g. with 'pybind11>=2.12'.
> 
> If you are a user of the module, the easiest solution will be to
> downgrade to 'numpy<2' or try to upgrade the affected module.
> We expect that some modules will need time to support NumPy 2.
>

adam-urbanczyk, 2024-07-10:
How did you install? Via pip or conda?

drayde, 2024-07-10:
Pip.

Install log here:
https://github.com/cadscriptHQ/cadscript/actions/runs/9879762014/job/27286700947#step:6:1599

adam-urbanczyk, 2024-07-10:
Then I think you need to ask the maintainers of nlopt to build against numpy 2 and release.

drayde, 2024-07-10:
Yes, I think you are right. I'll use numpy<2 for now

PR #1617 (2024-06-28): Get rid of nptyping
Nptyping does not seem to be actively maintained and is not strictly needed.

Closes #1616 and https://github.com/CadQuery/CQ-editor/issues/442