CadQuery/cadquery #1618: is there a way to sketch on other than the XY plane without being limited to a face on an object?

MotoCyti, 2024-06-28:
running cadquery 2.4.0
mysketch = cq.Sketch().rect . . . .  is always on the YX plane.  But I really need to sketch on the YZ plane.  
The sketch isn't the basis for any objects, it's to layout construction points, lines, and circles.
I can stick it to the YZ plane if I start it like this:
mysketch = cq.Workplane('YZ').sphere(.001).faces('>X').sketch().circle(0.5).rect(2,6)...
but that is using "sketch()" as opposed to "Sketch" and some of the Sketch operations are not available on sketch.  Notably rarray.
But with that said, Sketch().rarrray(0.6,1,5,1) doesn't appear to work anyhow, so it's perhaps a moot point.
I also can't figure how to draw a circle with sketch that is other than at the origin.  No luck with offset, moved, or located.  
This is all perhaps a moot put, but I though you might be interested to know that 1) I think there is something fishy about rarray, and 2) (not your problem, but FYI) Sketch/sketch is a bit obtuse.  I'm going to do my "2D" contruction sketch using regular 3D objects.

--Chris

lorenzncode, 2024-06-29:
You can create a sketch on `Workplane` like this: 

```py
import cadquery as cq

s1 = (
    cq.Sketch()  # create Sketch
    .rect(100, 200)  # create rect, the default mode is additive
    .circle(20, "s")  # s is for subtractive mode
)

w1 = (
    cq.Workplane("YZ")  # YZ plane
    .placeSketch(s1)  # place the sketch
    .extrude(15)  # extrude in plane normal direction
)
```

If you mean to select say a wire or face from the Sketch there is also #1597.

There are two `Workplane` methods that integrate Sketch. 
https://cadquery.readthedocs.io/en/latest/sketch.html#workplane-integration

`placeSketch`: this is used to create the sketch as a separate entity and then placed on the Workplane
`sketch`: to create the sketch inline in one fluent chain of calls

When using the inline form, you must call `finalize` to end the sketch.

```py
r = cq.Workplane("YZ").sketch().rect(100, 200).circle(20, "s").finalize().extrude(15)
```

All sketch methods such as `rarray` are available independent of the Workplane integration method you choose.

```py
r = cq.Workplane("YZ").sketch().rarray(20, 0, 4, 1).rect(10, 5).finalize().extrude(2)
```

> to draw a circle with sketch that is other than at the origin

Try `push`:

```py
r = (
    cq.Workplane("YZ")
    .sketch()
    .rect(100, 200)
    .push([(0, 50)])
    .circle(20, "s")
    .finalize()
    .extrude(15)
)
```

MotoCyti, 2024-06-30:
Wonderful.  Thank you sooo much.

closed 2024-06-30 by MotoCyti