CadQuery/cadquery #1619: Workplane translate does not move the tags

lenianiva, 2024-07-01:
Consider this example which should produce 3 cubes stacked on top of each other:
```python
import cadquery as Cq

def tagged_box(tag=""):
    result = (
        Cq.Workplane('XY')
        .box(10, 10, 10)
    )
    result.faces("<Z").tag(f"{tag}bot")
    result.faces(">Z").tag(f"{tag}top")
    return result

def tagged_double_box():
    result = (
        tagged_box("b1_")
        .union(
            tagged_box("b2_")
            .translate((0, 0, 5))
            .rotate((0, 0, 0), (0, 0, 1), 45)
        )
    )
    return result

result = (
    Cq.Assembly()
    .add(tagged_box(), name="parent", color=Cq.Color("blue1"))
    .add(tagged_double_box(), name="child", color=Cq.Color("yellow1"))
    .constrain("parent?bot", "child?b2_top", "Plane")
    .solve()
)

show_object(result)
```
but instead the mating surface `child?b2_top` stays in place, despite that the object was moved up by 5mm.

[image]

Is this intended behaviour? How can I get the tag `b2_top` to move during `.translate()`?

lorenzncode, 2024-07-03:
Methods such as `translate` create new `Workplane` objects in the chain.  

See also docs introspective example for more.  It should help with tags usage. https://cadquery.readthedocs.io/en/latest/primer.html#an-introspective-example:
_Returning a new instance of Workplane is the normal behaviour of most Workplane methods...
Tags offer a way to refer back to a previous Workplane_

lenianiva, 2024-07-03:
> Methods such as `translate` create new `Workplane` objects in the chain.
>
> See also docs introspective example for more. It should help with tags usage. [...]

Is there a way to move all objects in a workplane while preserving the tags?

lorenzncode, 2024-07-03:
Yes, it's possible.  There is no built-in method, but can be done as follows:
1. Iterate over `<the Workplane>.ctx.tags.values()`
2. Directly update each tagged Workplane object list (translate each object)

closed 2024-07-17 by lenianiva