CadQuery/cadquery #1620: Add materials to assemblies

adrianschneider94, 2024-07-01:
As far as I can see, it's only possible to add colors so far.
That could be extended to handling materials by using `XCAFDoc_Material` for physical properties and `XCAFDoc_VisMaterial` for visualization.

I implemented a first working approach:
https://github.com/sangl-spezialtransformatoren/cadquery/tree/materials

The syntax is at the moment:

```python

part = create_some_part()
assembly = Assembly()

copper = Material(
    name="Copper",
    base_color=Color(0.955, 0.637, 0.538, 1),
    density=8.9,
    metalness=1,
    roughness=0.1,
)

assembly.add(part, material=copper)

```

I wanted to use this issue as a discussion entrypoint, a pull request could follow later.

Links to OCCT documentation:
* https://dev.opencascade.org/doc/refman/html/class_x_c_a_f_doc___material.html
* https://dev.opencascade.org/doc/refman/html/class_x_c_a_f_doc___vis_material.html

adam-urbanczyk, 2024-07-02:
This would be interesting. What is your main use case?

adrianschneider94, 2024-07-03:
One is mass calculation, where the density is necessary. The other is, that you can export with the visualization settings, for example you can visualize metal nicely with gltf.

Jopie01, 2024-07-03:
I'm also interested in this. Maybe it's also possible to add textures? My main case is that I'm creating a structure and want to create an (rendered, would it be possible?) image of it to be added to a printed document. Also the 3D model is exported for visualization in a viewer.

lenianiva, 2024-07-03:
This would be very useful for calculating load!

adrianschneider94, 2024-07-04:
@Jopie01 Textures should be possible, [XCAFDoc_VisMaterialPBR](https://dev.opencascade.org/doc/refman/html/struct_x_c_a_f_doc___vis_material_p_b_r.html) has fields for that, but I don't know yet, how those work.

adam-urbanczyk, 2024-07-04:
Alright, let's give it a try @adrianschneider94 ! I did take a look at your branch and AFAICT you did some reformatting/reordering of imports. I will kindly ask to revert this. Our README contains instructions regarding which black version to use. Please do not apply other formatting fixes.

Regarding the PR I see the following tasks. First two are MVP, the rest would be nice to have but can ba via other PRs.

- [ ] Add the vocabulary type `cq.Material`
- [ ] Update Assembly API
- [ ] Update toCAF to write VisMaterial and Material 
- [ ] STEP export (?)
- [ ] (de)serialization via pickle (?)

adrianschneider94, 2024-07-04:
Yes, so far I have not set up everything, just got everything working for my project.

Some points:
* I adapted the approach to keep all data in OCP objects using 'wrapped' attributes. But as for example `XCAFDoc_VisMaterial.PbrMaterial()` won't return a ref but a copy of the struct, I now think it's easier to define a python class holding the material properties and then apply them in `toCAF()`. What do you think about that?
* Exporting STEP and GLTF works out of the box (the other formats too I guess), but as far as I know, STEP does not have material properties.
* Regarding serialization: Is there already some approach in the library to serialization yet? I haven't seen it yet and even started to write own wrappers using pydantic. Would be great if that's already implemented.

adam-urbanczyk, 2024-07-05:
[I updated your comment with numbers to be able to refert to the points]

[I updated your comment with numbers to be able to refer to the points]

Alright, here are my answers

1. I did no analyze in detail, but that approach sounds OK too
2. Indeed GLTF should work via CAF regarding vis props. IDK about material props. STEP does not support vis props other than color, but maybe we could encode material props in some way
3. Only serialization to CAF is there. No deserialization ATM. For unrelated reasons I want to (finally!) work on pickling. It should solve it.

Long story short, I’d propose to keep the scope small and for now “just” implement materials up to the CAF part.

pddees, 2025-03-20:
Has there been any movement on this? I'd also like to use this feature for loads, and generating geometries with nice textures

adrianschneider94, 2025-03-21:
Not yet, but I have this on my task list for the comming weeks.

adrianschneider94, 2025-04-15:
I created a pull request #1815.