CadQuery/cadquery #1626: Attributeerror for bool8 to bool in numpy?

dead-by-noon, 2024-07-12:
[image]

Hi all, with just a basic installation of cadquery, im getting this issue:

raise AttributeError("module {!r} has no attribute "
AttributeError: module 'numpy' has no attribute 'bool8'. Did you mean: 'bool'?

my numpy version is 2.0.0, do I need to downgrade it in order to use cadquery?

jmwright, 2024-07-12:
See #1616

sbalk, 2024-08-21:
For now, downgrade numpy to 1.23.5 because from [1.24 it's deprecated](https://numpy.org/doc/stable/release/1.24.0-notes.html#np-str0-and-similar-are-now-deprecated).

AnonymouZEE, 2024-08-31:
same problem... 
please replace all numpy.bool8 with numpy.bool_ 
...or avoid NumPy types for booleans

adam-urbanczyk, 2024-09-01:
This is fixed in master, see #1617

YahliGilboa, 2024-09-06:
still didnt work for me. had to revert to numpy 1.23.5

adam-urbanczyk, 2024-09-06:
What did not work exactly in master? How did you install it? Which dep gives the error?

ulikoehler, 2024-09-12:
I can confirm that upgrading to cadquery master fixes this issue for me.

Command I used:

```sh
pip install --user -U git+https://github.com/CadQuery/cadquery.git --break-system-packages
```

with current master revision = [5429f61](https://github.com/CadQuery/cadquery/commit/5429f618c6723b47f07ba5394a912e22b0dd5cff)

zakzook333, 2024-09-13:
Neither solution worked for me, I always get "Process finished with exit code -1073741819 (0xC0000005)" error" and I don't know why. I tried clean installation from master in virtual environment with latest pip, but I still get this error.

ulikoehler, 2024-09-13:
@zakzook333 I believe the "Process finished fwith exit code" is a totally different error and I'd guess its opencascade related but not sure. Can you share a full stacktrace?

zakzook333, 2024-09-13:
I don't have a full stracktrace. I only put "import cadquery" into the script and it gives me this error. The interesting thing is that when I type "import cadquery" followed by "cadquery.Workplane('XY').box(1,2,3).toSvg()" directly into python console I get raw SVG output as I am supposed to get. I tried to run the script from both python console and PyCharm.

davidedelvento, 2024-11-23:
> AttributeError: 'Editor' object has no attribute 'run_cell_action'

A decent workaround is to run the following. This is assuming you have a dedicated virtual environment for this, if you don't, you are likely to break other things. If you use conda/mamba, I have no idea, since I don't use them

```
pip install nlopt==2.7.1
pip install numpy==1.23.5
pip install spyder==5
```

Explanation:
* the numpy is what is suggested to do by others in this ticket, however that alone is insufficient, because it conflicts with the latest `nlopt` which is otherwise installed, so we need to downgrade that
* such a downgrade is a pain to identify but you can get a list of candidates with `pip index versions nlopt`. Then by trial and error which each version, go into a browser and try every version starting from https://pypi.org/pypi/nlopt/2.9.0/json and search for the numpy dependency. The newest version that does not require numpy 2 is https://pypi.org/pypi/nlopt/2.7.1/json and that is what I selected
* With that solved, we are bitten by https://github.com/CadQuery/CQ-editor/issues/455 and the last line solves that

But yes, it would be nice to have such things included in the `requirements.txt` rather than having to reverse engineering it.

jmwright, 2025-01-07:
This should be fixed now that CadQuery 2.5.2 is available.

closed 2025-01-07 by jmwright