CadQuery/cadquery #1632: Sketches edges connection misplaced

eddieliberato, 2024-07-16:
When I run 2 of the sketches from the docs on my locally ( cadquery 2.5.0.dev0 / cadquery-ocp 7.7.2) I get shapes that differ from the ones on the docs.

[image]
[image]

adam-urbanczyk, 2024-07-16:
I do not recognize the tool. Can you reproduce using `cadquery.vis.show`?

eddieliberato, 2024-07-18:
Hey Adam, 

> I do not recognize the tool. Can you reproduce using `cadquery.vis.show`?

I didn't know this trick :joy: . Very cool ! Nice to learn this. 

I did a bit of further investigation and actually I am now assuming it is somehow related to the step exporter. 
Running it using vis.show returns exactly what's expected. 

[image]

Exporting it to step, importing again and running vis.show on the imported file, shows that weird edge, however. 

[image]

Stl works fine. 

Can you reproduce this? 

P.s.
Sorry for being so brief last time, I was under a bit of time pressure. 

I recently got  a new desktop and now I am running Arch / Wayland. My toolchain at the moment for 3d visualization is [mayo](https://github.com/fougue/mayo). I modified the source code so it auto reloads every time an open file changes in disk without any interruption. This allows me not having to leave my doom emacs setup to edit code. I got too attached to vim keybinds I guess. :sweat_smile:

adam-urbanczyk, 2024-07-18:
I can reproduce the issue after a step export/import round-trip. Brep export/import works fine.

adam-urbanczyk, 2024-07-18:
BTW: as a workaround you can also use CQ-editor with autoreload enabled.

eddieliberato, 2024-07-18:
> BTW: as a workaround you can also use CQ-editor with autoreload enabled.

gotcha. 

I was just having a look on the issue now and found [this](https://dev.opencascade.org/content/importing-geometry-fails-step-works-brep) which seems to be related because making the outer triangle a bit bigger, the step translation works fine. 

From the [OCCT](https://dev.opencascade.org/doc/overview/html/occt_user_guides__step.html) documentation we see this: 

> Setting the translation parameters
> 
> The following parameters are used for the OCCT-to-STEP translation.
> >write.precision.mode
> 
> writes the precision value.
> 
> Least (-1) : the uncertainty value is set to the minimum tolerance of an OCCT shape
> Average (0) : the uncertainty value is set to the average tolerance of an OCCT shape.
> Greatest (1) : the uncertainty value is set to the maximum tolerance of an OCCT shape
> Session (2) : the uncertainty value is that of the write.precision.val parameter.
> 
> Read this parameter with:
> 
> Standard_Integer ic = Interface_Static::IVal("write.precision.mode"); Modify this parameter with:
> if(!Interface_Static::SetIVal("write.precision.mode",1))  
> .. error .. 
> 
> Default value is 0.
>> write.precision.val
> 
> a user-defined precision value. This parameter gives the uncertainty for STEP entities constructed from OCCT shapes when the write.precision.mode parameter value is 1.
> 
> 0.0001: default
> any real positive (non null) value.

As I understood on cadquery we can pass precision_mode as a parameter but not precision_val, however.

adam-urbanczyk, 2024-07-19:
In princple you can do (almost) anything using OCP:
```python
import cadquery as cq
import OCP

assert OCP.Interface.Interface_Static.SetRVal_s("write.precision.val",1e-9)
assert OCP.Interface.Interface_Static.SetIVal_s("write.precision.mode",2)

result = (
    cq.Workplane()
    .box(5, 5, 1)
    .faces(">Z")
    .sketch()
    .regularPolygon(2, 3, tag="outer")
    .regularPolygon(1.5, 3, mode="s")
    .vertices(tag="outer")
    .fillet(0.2)
    .finalize()
    .extrude(0.5)
)

cq.exporters.export(result,'out.step')
result = cq.importers.importStep('out.step')
```
This does not solve the issue though.

Jopie01, 2024-07-23:
Sorry for the noise ..... but

> .... for 3d visualization is [mayo](https://github.com/fougue/mayo). I modified the source code so it auto reloads every time an open file changes in disk without any interruption.

I also use Mayo because I can do some measuring and clipping. Do you mind to share the changes?

eddieliberato, 2024-07-29:
Hey Jopie,

Sorry for the late reply. 

You may wanna have a look at [this issue I just replied on their repo](https://github.com/fougue/mayo/issues/279). It as a dirty hack but it works for me. 

Perhaps you can also make some noise there so they hear us 😂😂

All the best, 
Edi