CadQuery/cadquery #1636: Visualizing multiple vertices (3d points)

HarmenJorritsma, 2024-07-22:
Very nice and interesting library. I am a newbie, and I like to learn this library.

As a challenge, I like to model (complex) polyhedrons. Therefore I want firstly visualize the vertices (3d points). I tried this script:

```python
verticesXYZ = (
    (0,0,0), (1,0,0), (1,1,0),(0,1,0),
    (0,0,1), (1,0,1), (1,1,1),(0,1,1)
)

vertices = []

for v in verticesXYZ:
    vertices.append(cq.Vector(v))

show(vertices)
```

This code results in an NotImplementedError for the list of Vectors I put into the show function. I also tried to make objects of class cq.Vertix instead of cq.Vector. Showing Vertix objects throws the following error: tuple' object has no attribute 'IsNull'". I use Jupyter Cadquery with Visual Code as GUI. Furthermore I tried to put the vertices in a Workplane, Sketch or Assembly. Other objects like cubes etc, are showed correctly. Showing 1 point is also working. 

Any suggestions to show multiple points? 

Kind regards!

p.s. I am also interested in more conceptual background documentation/literature about the concepts of CADQuery and 3d modelling in common. Sharing links is appreciated!

adam-urbanczyk, 2024-07-23:
You can display vertices in CQ-editor. I do not know about other tools, you'll need to open issues on their trackers and not here.

```python
from cadquery.occ_impl.shapes import *

verticesXYZ = (
    (0,0,0), (1,0,0), (1,1,0),(0,1,0),
    (0,0,1), (1,0,1), (1,1,1),(0,1,1)
)

vertices = []

for v in verticesXYZ:
    vertices.append(vertex(*v))

show_object(vertices)
```
[image]

HarmenJorritsma, 2024-07-24:
Thanks for reply! Visualizing in cq-editor works also for me. So the issue is a problem with Jupyter Cadquery for visualizing vertices.