CadQuery/cadquery #1639: free function sweep BRepOffsetAPI_MakePipeShell  option to SetMode

lorenzncode, 2024-07-27:
I did not expect this result from the sweep.

```py
import cadquery as cq
from cadquery.occ_impl.shapes import *

path = cq.Wire.makeHelix(2, 6, 4)

e1 = segment((4, 0, 0), (4, 0, 0.5))

res = sweep(e1, path)
```

[image]

I got the expected result after changing the sweep function to call `builder.SetMode(True)` here:
https://github.com/CadQuery/cadquery/blob/3cd327a2dbaf3bae6791463a3dcb0ad7a714b607/cadquery/occ_impl/shapes.py#L4956

[image]

Perhaps this can be set by default, or optionally as a function argument.

lorenzncode, 2024-11-10:
If I understand the new sweep `aux` param usage correctly, this is supported now in #1692.

```py
import cadquery as cq
from cadquery.occ_impl.shapes import *

path = cq.Wire.makeHelix(2, 6, 4)

e1 = segment((4, 0, 0), (4, 0, 0.5))

res = sweep(e1, path, segment((0, 0, 0), (0, 0, path.endPoint().z)))
```

adam-urbanczyk, 2024-11-11:
Solved by #1692

closed 2024-11-11 by adam-urbanczyk