CadQuery/cadquery #1640: Operation along a rounded box

jniclas, 2024-08-04:
Hey, I have a box which is rounded, now I want to select the upper edge (all around) and want to apply an operation every e.g. 2cm. Like in this picture:
[image]
How can I perform this operation? I have already:

```python
length = 180
width = 100
height= 80
wall = 0.4*2
topOffset = 5

part = (
    cq.Workplane("XY")
    .rect(width, length)
    .workplane(offset=height)
    .rect(width + topOffset, length + topOffset)
    .loft(ruled=True)
    .edges("not(|X)").edges("not(|Y)")
    .fillet(3)
)

braceWidth = wall
braceDistanceAbsolute = 20

brace = (
    cq.Workplane("XZ")
    .lineTo(0, -height)
    .lineTo(-topOffset, -height)
    .close()
    .extrude(wall)
)

circlePath = part.edges(">Z and >X").val()
print(circlePath.Length())

braceDistance = (braceWidth + braceDistanceAbsolute) / circlePath.Length()
braceCount = floor(circlePath.Length()/(braceDistanceAbsolute+braceDistance))

print(braceDistance)
print(braceCount)

for i in range(braceCount):
    relative_position_on_path = i*braceDistance
    wire_tangent = circlePath.tangentAt(relative_position_on_path)
    wire_angle = degrees(atan2(wire_tangent.y, wire_tangent.x))
    part = part.union(brace.rotate((0,0,0),(0,0,1), wire_angle-90).translate(circlePath.positionAt(relative_position_on_path)))

```
only issue is, that `circlePath` is not the full loop, but only one side 🤔

lorenzncode, 2024-08-04:
To select the outer wire on the top face try:
```py
circlePath = part.wires(">Z").val()
```

adam-urbanczyk, 2024-08-04:
You could use `apply` and `cq.Wire.positionAt`, e.g.
```python
N = 12

res = (
    cq.Workplane()
    .box(1,1,1)
    .wires('>Z')
    .apply(lambda x: list(x[0].positionAt(p/N) for p in range(N)))
    .sphere(0.1, combine='cut')
)
```