CadQuery/cadquery #1641: Length of an edge

lenianiva, 2024-08-04:
Is there a way to get the length for an `Edge` object? In particular I'm interested in the length of splines and bezier curves.

adam-urbanczyk, 2024-08-04:
You can use `cq.Edge.Length`

lenianiva, 2024-08-04:
> You can use `cq.Edge.Length`

I don't see that function here: https://github.com/CadQuery/cadquery/blob/3cd327a2dbaf3bae6791463a3dcb0ad7a714b607/cadquery/occ_impl/shapes.py#L1920. Where is it defined?

adam-urbanczyk, 2024-08-04:
https://github.com/CadQuery/cadquery/blob/3cd327a2dbaf3bae6791463a3dcb0ad7a714b607/cadquery/occ_impl/shapes.py#L1760

lenianiva, 2024-08-04:
> https://github.com/CadQuery/cadquery/blob/3cd327a2dbaf3bae6791463a3dcb0ad7a714b607/cadquery/occ_impl/shapes.py#L1760

Thanks. I didn't see it in the documentation so I was wondering about this

closed 2024-08-04 by lenianiva

adam-urbanczyk, 2024-08-04:
Yup, probably sphinx does not process mixins.