CadQuery/cadquery #1643: Workplane.clone()

wirekang, 2024-08-10:
I know I can define a function or use `w1.translate((0,0,0))` to create a new reference, but it would be great if there is a simple `clone` method:

```python
w1 = cq.Workplane().box(1,2,3)
w2 = w1.clone()

w2.add(...)
```

```python
    def clone(self: T) -> T:
        return self.newObject([*self.objects])
```

adam-urbanczyk, 2024-08-11:
What is the use case? I.e. are you after a deep or shallow copy in the end?

wirekang, 2024-08-11:
I thought CadQuery is based on immutable objects, since there are chaining methods with description `Returns a copy of...`. So the use case I thought was like:

```python
left_pin = cq.Workplane()...
right_pin = left_pin.translate((0,0,0)) // Oh this is so hacky
cq.Assembly().add(left_pin).add(right_pin)...
```

You know this is wrong. They shares many things like CQObject and CQContext. If I understand correctly, `newObject` has nothing to do with immutable but for parent stack.

So if there is a `clone` method, it should deep copy everything, but I can't sure this kind of approche is valid in CadQuery.