CadQuery/cadquery #1650: Which center fucntion is more appropriate for "Point constraint", Center()(=centerOfMass)  or CenterOfBoundBox()?

huskier, 2024-08-22:
In the CQ's document, it is said that "The Point constraint uses the Center() to find the center of the argument. ". 

In some cases, the centerOfMass is the same as the CenterOfBoundBox. However, in some cases, they are not the same. For example, assume we have the 3D text of "CADQUERY", the Center() of the 3D text is not the same as the CenterOfBoundBox. The following is the example code. So, here is the question: which center fucntion is more appropriate for "Point constraint", Center()(=centerOfMass)  or CenterOfBoundBox()?

```
import cadquery as cq

text_string = "CadQuery"
font = "Arial"
size = 10
text3D = cq.Workplane("XZ").text(text_string, font=font, fontsize=size, distance=2) 

print("text3D.val().BoundingBox().center is ", text3D.val().BoundingBox().center)
print("text3D.val().Center is ", text3D.val().Center())
```

lorenzncode, 2024-08-22:
Here is another example of an object where center of mass differs from center of bound box.  Perhaps you can create a dummy vertex if you choose to create the constraint based on bound box.

```py
import cadquery as cq
from cadquery.occ_impl.shapes import *

sk1 = cq.Sketch().rect(20, 10)
sk2 = cq.Sketch().circle(4).moved(5)
# the object with center that differs from center of bound box
obj1 = cq.Workplane().placeSketch(sk1 - sk2).extrude(1)

sph = sphere(1)

assy = cq.Assembly(name="assy")
assy.add(obj1, name="obj1")

# two spheres as markers
assy.add(sph, name="sph1", color=cq.Color("red"))
assy.add(sph, name="sph2", color=cq.Color("green"))

assy.constrain("obj1", "Fixed")

# the red sphere at face center of mass
assy.constrain("obj1@faces@>Z", "sph1", "Point")

# green sphere at face center of bound box
assy.constrain(
    "obj1", vertex(obj1.faces(">Z").val().CenterOfBoundBox()), "sph2", sph, "Point"
)

assy.solve()
```

[image]

huskier, 2024-08-23:
@lorenzncode Thanks for your explanation.

closed 2024-08-23 by huskier