CadQuery/cadquery #1652: When the same sketch is placed on the "XZ-plane" and the "ZX-plane", how to keep the sketch's direction the same?

huskier, 2024-08-24:
When we put the the same sketch on the "XZ-plane" and the "ZX-plane" respectively, the sketch on the "XZ-plane" is one direction, 
and the sketch on the "ZX-plane" is another direction (see the code below). However, we want to get the same sketch direction no mather whether the plane is the "XZ-plane" or the "ZX-plane". Could we get the result with the current APIs? Or could we get the result through manipulating the sketch when we place the sketch on the workplane?

```
import cadquery as cq
from cadquery.vis import show

rect_sketch = cq.Sketch().rect(10, 6.18)

show(rect_sketch)

sketch_on_XZ_plane = (
    cq.Workplane("XZ")
    .rect(20, 20)
    .extrude(2)
    .placeSketch(rect_sketch)
    .cutThruAll()
)

show(sketch_on_XZ_plane)

sketch_on_ZX_plane = (
    cq.Workplane("ZX")
    .rect(20, 20)
    .extrude(2)
    .placeSketch(rect_sketch)
    .cutThruAll()
)

show(sketch_on_ZX_plane)

```

adam-urbanczyk, 2024-08-24:
On master you could try this: `.placeSketch(rect_sketch.moved(rz=90))`

huskier, 2024-08-25:
Yes, the new improved Sketch.moved(...) function has solved my problem. I appreciate the continuous evolution of CQ! Thank you very much for your effort.