CadQuery/cadquery #1654: AttributeError: 'Workplane' object has no attribute 'cylinder'

WWWladWWW, 2024-08-30:
In cadquery 2.4.0, I get  `AttributeError: 'Workplane' object has no attribute 'cylinder'` when i running this simple programm in CQ-editor:

```
import cadquery as cq
# Create the cylinder
cylinder = c =cq.Workplane().cylinder(5,1)
# Display the result
show_object(cylinder)
```
Meanwhile if i change attribute to `box` it works just fine:

```
import cadquery as cq
# Create the cylinder
cylinder = c =cq.Workplane().box(5,1,1)
# Display the result
show_object(cylinder)
```

What do i do wrong?

jmwright, 2024-08-31:
I just set up a CadQuery 2.4.0 environment and cannot reproduce this error. Are you sure that the package of CQ-editor you are running has CQ 2.4.0 or newer bundled? You can check for sure by typing the following in the CQ-editor console.
```python
import cadquery
print(cadquery.__version__)
```

adam-urbanczyk, 2024-08-31:
Something wrong with your install? Cylinder is definitely there in 2.4: https://github.com/CadQuery/cadquery/blob/c44978d60cee2d61bdadf4cb4498286b7034b4c6/cadquery/cq.py#L4023

WWWladWWW, 2024-08-31:
> I just set up a CadQuery 2.4.0 environment and cannot reproduce this error. Are you sure that the package of CQ-editor you are running has CQ 2.4.0 or newer bundled? You can check for sure by typing the following in the CQ-editor console.
> 
> ```python
> import cadquery
> print(cadquery.__version__)
> ```

Yes indeed it was a CadQuery 2.1 bundled in CQ-editor 0.2 i downloaded from here:
https://github.com/CadQuery/CQ-editor/releases/tag/0.2
Downloading nightly built compleatly resolved issue. Thank you!
https://github.com/CadQuery/CQ-editor/releases/tag/nightly

closed 2024-08-31 by WWWladWWW