CadQuery/cadquery #1660: Wires selection logic of "cq.NearestToPointSelector((0, 0, 1)) or cq.NearestToPointSelector((20, 0, 1))" does not work.

huskier, 2024-09-04:
There are some cutouts on a panel. We want to select the useful wires with the selection logic of "cq.NearestToPointSelector((0, 0, 1)) or cq.NearestToPointSelector((20, 0, 1))", but it does not work as expected. The following is the code. What's the approriate approach for such operation?

```
import cadquery as cq
from cadquery.vis import show

panel = cq.Workplane().box(60, 40, 2)

panel_with_cutouts = panel.faces(">Z").workplane().rect(29, 21).moveTo(-20, 0).circle(2.5).moveTo(20, 0).circle(2.5).cutThruAll()
show(panel_with_cutouts)

new_obj0 = panel_with_cutouts.faces(">Z").wires(cq.NearestToPointSelector((20, 0, 1)))
show(new_obj0)

new_obj1 = panel_with_cutouts.faces(">Z").wires(((cq.NearestToPointSelector((0, 0, 1))) or (cq.NearestToPointSelector((20, 0, 1)))))
show(new_obj1)
```

lorenzncode, 2024-09-05:
Try `SumSelector`?

```py
new_obj2 = panel_with_cutouts.faces(">Z").wires(
    cq.selectors.SumSelector(
        cq.selectors.NearestToPointSelector((0, 0, 1)), cq.selectors.NearestToPointSelector((20, 0, 1))
    )
)
```

lorenzncode, 2024-09-06:
My bad example - use it with the overload instead.  See the `__add__` overload. 

```py
new_obj3 = panel_with_cutouts.faces(">Z").wires(
    cq.selectors.NearestToPointSelector((-20, 0, 1))
    + cq.selectors.NearestToPointSelector((0, 0, 1))
    + cq.selectors.NearestToPointSelector((20, 0, 1))
)
```

lorenzncode, 2024-09-06:
For the `BoxSelector` wire selection, try:

```py
new_obj2 = panel_with_cutouts.faces(">Z").wires(
    cq.selectors.BoxSelector((-25, 12, 2), (25, -12, 0), True)
)
```
See https://github.com/CadQuery/cadquery/blob/6d7d1d640d87410913f81b96babf914fd45f77a3/cadquery/selectors.py#L144

huskier, 2024-09-06:
@lorenzncode Thank you for the updates.

Now, the BoxSelector selection result is as expected.
```
new_obj2 = panel_with_cutouts.faces(">Z").wires(
    cq.selectors.BoxSelector((-25, 12, 2), (25, -12, 0), True)
)
```
[image]

However, the NearestToPointSelector __add__ overload selection gives unexpected results.
```
new_obj3 = panel_with_cutouts.faces(">Z").wires(
    cq.selectors.NearestToPointSelector((-20, 0, 1))
    + cq.selectors.NearestToPointSelector((0, 0, 1))
    + cq.selectors.NearestToPointSelector((20, 0, 1))
)
```
[image]

Then, I've put a line of debugging code into the NearestToPointSelector class:
[image]
The debugging info is as following.
```
tShape's type is  <class 'cadquery.occ_impl.shapes.Wire'>
```
However, Wire shape has no Center() function. I do not know whether it is related to the unexpected result.

I've also put the debugging code in Shape's Center() function(). And the Shape's Center() function is called. No more any digging for me.
```
In Shape's Center() function, type(self) is  <class 'cadquery.occ_impl.shapes.Wire'>
```
[image]

lorenzncode, 2024-09-08:
> Therefore, cq.selectors.NearestToPointSelector((0, 0, 1)) can not guarantee to select the target wire.

Yes, agree that if two or more objects share the same center, then `NearestToPointSelector` is not sufficient on its own.   It's not a bug.  It's a case where a different selector is more appropriate or you might chain with another selector.

huskier, 2024-09-08:
@lorenzncode  Thank you for your help.

closed 2024-09-08 by huskier