CadQuery/cadquery #1662: Segfault when operating on imported geometry

arsenixprime, 2024-09-07:
I am using cadquery to make some support scripts to assist with making drawer organizers (for storing tools). I have functions to add keyed features after the organizers are split up into pieces sized for 3d printing. Some of the models I made natively in cadquery. Others that are more complex I modeled in Solidworks, exported as STEP, and brought into cadquery for adding joining features. This works on most models but on a more complex one cadquery segfaults immediately on any geometry operation even if the resulting new geometry doesn't even intersect the imported geometry. All the files import successfully without errors and can be reexported as STLs without modification.

There is no debugging information so I'm not really sure where to begin. Two of my parts are bringing out this fault. They are very different but have some similar geometry.

## To Reproduce
3 files are attached in a zipfile.
* DrawerModule_Caliper1.STEP - Works fine
* DrawerModule_Mics1.STEP - Segfault
* DrawerModule_Mics2.STEP - Segfault

## Backtrace

```
```

## Environment

OS:

Was CadQuery installed using Conda?: 
Output of `conda list` from your active Conda environment:
```
```

Using:

arsenixprime, 2024-09-07:
Happy to do some debugging on this. For some reason it just does not seem to like the geometry of these models. Thank you!

arsenixprime, 2024-09-07:
I paired the geometry down until it successfully worked. It appears that the sketched radius on the rotated cuts is what is causing the segfault. When I changed these into chamfers the parts can be operated on successfully.
[image]

arsenixprime, 2024-09-07:
After going back to full part I had to eliminate all fillets on these rotational cuts even ones that were applied after the sketch. Both convex and concave.
[image]

This certainly seems like some sort of bug. I believe this type of geometry can be created with cadquery.

adam-urbanczyk, 2024-09-12:
The geometry looks reasonable, do you happen to have a backtrace?