CadQuery/cadquery #1665: One line that simply crashes all RAM memory

filaPro, 2024-09-11:
```
import cadquery as cq
cq.Workplane().sketch().push([(48.5800789794081,-3.3737790291878085)]).rect(3.0,87.6847679284009).rect(3.0,88.65465934289723).polygon([(0,0),(1,1),(2,2)],mode="s").finalize().extrude(1)
```

cadquery==2.4.0 is installed throw pip

jmwright, 2024-09-11:
Please show the output of `pip list` and tell us which OS you're on.

filaPro, 2024-09-11:
thank you for quick response. i'm inside docker with ubuntu 22.04.3. the host os outside docker is ubuntu 20.04.6

```
Package                   Version
------------------------- --------------
accelerate                0.31.0
addict                    2.4.0
aiohttp                   3.9.5
aiosignal                 1.3.1
anaconda-anon-usage       0.4.4
antlr4-python3-runtime    4.9.3
anyio                     4.4.0
archspec                  0.2.3
argon2-cffi               23.1.0
argon2-cffi-bindings      21.2.0
arrow                     1.3.0
asttokens                 2.0.5
astunparse                1.6.3
async-lru                 2.0.4
async-timeout             4.0.3
attrs                     23.1.0
Babel                     2.15.0
beautifulsoup4            4.12.3
bleach                    6.1.0
blinker                   1.8.2
boltons                   23.0.0
Brotli                    1.0.9
cadquery                  2.4.0
cadquery-ocp              7.7.2
casadi                    3.6.5
certifi                   2024.6.2
cffi                      1.16.0
chardet                   4.0.0
charset-normalizer        2.0.4
click                     8.1.7
comm                      0.2.2
conda                     24.5.0
conda-build               24.3.0
conda-content-trust       0.2.0
conda_index               0.5.0
conda-libmamba-solver     24.1.0
conda-package-handling    2.3.0
conda_package_streaming   0.10.0
ConfigArgParse            1.7
contourpy                 1.2.1
cryptography              42.0.5
cycler                    0.12.1
Cython                    3.0.10
dash                      2.17.1
dash-core-components      2.0.0
dash-html-components      2.0.0
dash-table                5.0.0
datasets                  2.20.0
debugpy                   1.8.2
decorator                 5.1.1
defusedxml                0.7.1
dill                      0.3.8
distro                    1.9.0
dnspython                 2.6.1
easydict                  1.13
einops                    0.8.0
exceptiongroup            1.2.0
executing                 0.8.3
expecttest                0.2.1
ezdxf                     1.3.1
fastjsonschema            2.20.0
filelock                  3.13.1
flash-attn                2.5.9.post1
Flask                     3.0.3
fonttools                 4.53.0
[... 142 lines cut ...]
tqdm                      4.66.4
traitlets                 5.14.3
transformers              4.42.3
trimesh                   4.4.9
triton                    2.3.1
truststore                0.8.0
types-dataclasses         0.6.6
types-python-dateutil     2.9.0.20240316
typing_extensions         4.11.0
typish                    1.9.3
tzdata                    2024.1
uri-template              1.3.0
urllib3                   2.2.2
wcwidth                   0.2.5
webcolors                 24.6.0
webencodings              0.5.1
websocket-client          1.8.0
Werkzeug                  3.0.3
wheel                     0.43.0
widgetsnbextension        4.0.11
xxhash                    3.4.1
yarl                      1.9.4
zipp                      3.19.2
zstandard                 0.22.0
```

filaPro, 2024-09-11:
if you are not able to reproduce this, i can try to make minimal reproducible docker image

jmwright, 2024-09-11:
It does not look like the polygon is closed, so the cause could be extruding an open/invalid sketch. If you remove the `polygon` call, and leave everything else, it works fine. One of the core devs who has worked more directly on the sketch feature can tell you more.

filaPro, 2024-09-11:
thank you again for your response. i also investigated that the problem is about the polygon with zero area. what i expect is to get an exception, but not like dying os due to out of memory. for example this line `cq.Worplane().sketch().polygon([(0,0),(1,1),(2,2)],mode="s").finalize().extrude(1)` produces `ValueError: Null TopoDS_Shape object`, and it is fine.

adam-urbanczyk, 2024-09-12:
Sometimes you'll get these kind of issues with invalid models. Here is a workaround:
```
import cadquery as cq
res = (
       cq.Workplane()
       .sketch()
       .push([(48.5800789794081,-3.3737790291878085)])
       .rect(3.0,87.6847679284009)
       .rect(3.0,88.65465934289723)
       .polygon([(0,0),(1,1),(2,2)],mode="s")
       .finalize()
       .extrude(1, clean=False)
)
```

Note that:
```
import OCP
OCP.BRepAlgoAPI.BRepAlgoAPI_Check(res.val().wrapped).IsValid()
```
results in
```
Out[8]: False
```

filaPro, 2024-09-13:
Thank you, i didn't know about this `clean=False`. Believe the memory leak will be fixed in future releases of cadquery or occ.