CadQuery/cadquery #1668: hole in model disappears when sides are filleted

bigfighter, 2024-09-20:
```
from cadquery.occ_impl.shapes import *

base = face(rect(50, 50)) - face(rect(40, 40))
base = base.fillet2D(5, base.vertices('<X')) # A
base = extrude(base, (0, 0, 5))
#base = base.fillet(0.5, base.edges('>X')) # B

from cadquery import version
print(version('cadquery'))

base # display in jypyter
```

if neither A nor B is commented out, then the hole disappears from the block. is this a bug?

[image]
[image]

I'm running cadquery in docker.

adam-urbanczyk, 2024-09-20:
Looks at least weird.

adam-urbanczyk, 2024-11-17:
It seems to be a bug of offset2D. Here is a workaround:
```
from cadquery.occ_impl.shapes import *

base = plane(50, 50)
base = base.fillet2D(5, base.vertices('<X')) - plane(40, 40) # A
base = extrude(base, (0, 0, 5))
```