CadQuery/cadquery #1671: cq.Solid.isSolid() not working?

bragostin, 2024-09-25:
The following code
```python
import cadquery as cq
part = cq.Workplane("XY").box(1,2,3)
test = cq.Solid.isSolid(part.val())
print("isSolid = ", test)
```
yields ```False```. Shouln't it yield ```True```?

adam-urbanczyk, 2024-09-25:
Confirmed, it's a bug

bragostin, 2024-09-26:
it looks like in 
```python
    def isSolid(obj: Shape) -> bool:
        """
        Returns true if the object is a solid, false otherwise
        """
        if hasattr(obj, "ShapeType"):
            if obj.ShapeType == "Solid" or (
                obj.ShapeType == "Compound" and len(obj.Solids()) > 0
            ):
                return True
        return False
```
```ShapeType``` should be ```ShapeType()```

adam-urbanczyk, 2024-09-26:
Yes, thanks, see #1672 .

closed 2024-10-08 by adam-urbanczyk