CadQuery/cadquery #1677: Show object of profile with multiple circles shows only 1

CasperH2O, 2024-10-05:
Hey there, I ran into the following oddity and I wonder if I am doing something incorrectly?

Version information:
CQ-Editor: 0.3.0dev
Python: 3.11
CADquery: 2.4.0

I have the following snippet to reproduce:

```
import cadquery as cq

# Create a circular profile (the inner and outer circles)
diameter_inner = 5
diameter_outer = 7

# Workplane on the XY plane
circle_profile = (
    cq.Workplane("ZX")
    .circle(diameter_outer / 2)  # Outer circle
    .circle(diameter_inner / 2)  # Inner circle
)

show_object(circle_profile)
```

However, the CQ models shows only one circle in the 3D view:

[image]

Am I missing the obvious or this a bug?

Fun fact, if I add a sweep on this model, the sweep shows up as expected, basically a tube shape:

```
import cadquery as cq

# Create a circular profile (the inner and outer circles)
diameter_inner = 5
diameter_outer = 7

# Workplane on the XY plane
circle_profile = (
    cq.Workplane("ZX")
    .circle(diameter_outer / 2)  # Outer circle (diameter 7)
    .circle(diameter_inner / 2)  # Inner circle (diameter 5)
)

show_object(circle_profile)

# Create a straight path for the sweep
path = cq.Workplane("ZY").lineTo(0, 20)

# Sweep the profile along the path
swept_shape = circle_profile.sweep(path)

# Display the result
show_object(swept_shape)
```

[image]

jmwright, 2024-10-05:
You have to call [consolidateWires](https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=consolidateWires#cadquery.Workplane.consolidateWires) first to get CQ-editor to display both circles.

CasperH2O, 2024-10-05:
Excellent, that did the trick @jmwright . 

For future reference, functional code:

```
import cadquery as cq

# Create a circular profile (the inner and outer circles)
diameter_inner = 8
diameter_outer = 10

# Workplane on the XY plane
circle_profile = (
    cq.Workplane("XZ")
    .circle(diameter_outer / 2)
    .circle(diameter_inner / 2)
    .consolidateWires()
)

show_object(circle_profile)
```

closed 2024-10-05 by CasperH2O