CadQuery/cadquery #1680: Discord invites expired

erooke, 2024-10-08:
All of the discord links in the readme are reporting that they are invalid.

jmwright, 2024-10-08:
@erooke There are only two Discord links that I see in the README, and the one that does not have `invite` in the URL is working for me. Have a look at #1681

jmwright, 2024-10-13:
@erooke Does [this link](https://discord.gg/Bj9AQPsCfx) work for you?

If not, are you using a VPN or anything that would keep these links from working?

jmwright, 2024-10-13:
@erooke I just now saw your comment below the image on the PR where you said using an incognito window fixed the issue. It sounds like a browser problem. Can you try another browser, check VPNs, etc?

closed 2024-10-13 by jmwright

erooke, 2024-10-14:
> @erooke I just now saw your comment below the image on the PR where you said using an incognito window fixed the issue. It sounds like a browser problem. Can you try another browser, check VPNs, etc?

Yeah, sorry about the false report. Still not sure what went wrong but they're all working for me now.