CadQuery/cadquery #1683: How to select a specific wire in a series of wires? The selection ".wire("<X")" works fine; however, the selection ".wires("<X[-2]")" refuses to work.

huskier, 2024-10-09:
The following is the MVP code. In the code, the selection ".wire("<X")" works fine; however, the selection ".wires("<X[-2]")" refuses to work at all. We want to select the wire just next to the wire ".wire("<X")". Any ideas?
```
import cadquery as cq
from cadquery.vis import show

panel = (
    cq.Workplane()
    .rect(200, 100)
    .extrude(2)
    .faces(">Z")
    .workplane(centerOption="CenterOfBoundBox")
    .pushPoints([(-80, 0), (-40, 0), (0, 0), (40, 0), (80, 0)])
    .hole(5)
)
show(panel)
show(panel.faces(">Z").wires(cq.selectors.LengthNthSelector(0)).wires("<X"))
show(panel.faces(">Z").wires(cq.selectors.LengthNthSelector(0)).wires("<X[-2]"))
```

lorenzncode, 2024-10-10:
Try `CenterNthSelector` (as string: ">>X[N]").

See the selector table here:
https://cadquery.readthedocs.io/en/latest/selectors.html#filtering-edges

```py
import cadquery as cq
from cadquery.occ_impl.shapes import *
from cadquery.vis import show

pts = [(-40, 0), (0, 0), (40, 0)]

sk1 = cq.Sketch().rect(100, 20).push(pts).circle(4, mode="s").reset()

face1 = sk1.faces().val()

edge0 = face1.edges("%CIRCLE").edges(">>X[0]")
edge1 = face1.edges("%CIRCLE").edges(">>X[1]")
edge2 = face1.edges("%CIRCLE").edges(">>X[2]")

label0 = text(">>X[0]", 8).translate(edge0.Center() + Vector(0, 16, 0))
label1 = text(">>X[1]", 8).translate(edge1.Center() + Vector(0, 16, 0))
label2 = text(">>X[2]", 8).translate(edge2.Center() + Vector(0, 16, 0))

show(face1, label0, label1, label2)
```
[image]

huskier, 2024-10-10:
Thank you very much, @lorenzncode. Your code solved our CQ's objects selection issue that had been bothering me for some time. All in all, edges are more friendly than wires in terms of CQ's objects selection.

adam-urbanczyk, 2024-10-10:
@huskier did you intend to reopen?

huskier, 2024-10-10:
@adam-urbanczyk Yes，Adam.
 I just wonder why we do not make the wires selectors as rich as the selectors of the faces and the edges? I think wires are more easy to understand compared to edges.

adam-urbanczyk, 2024-10-10:
In what sense are they less rich?

huskier, 2024-10-10:
For example, I expect the following statement works, and I think the statement make sense to me. Firstly, I've selected a series of wires. And then, I just want to select my wanted wire with the usual string selector("<X[-2]") which works for faces without problems.
`panel.faces(">Z").wires(cq.selectors.LengthNthSelector(0)).wires("<X[-2]")`

lorenzncode, 2024-10-12:
> I just want to select my wanted wire with the usual string selector("<X[-2]") 

This was discussed in #531.

">X[-2]" maps to `DirectionNthSelector` which "first filters for edges/faces parallel with the direction".

Given the filtering on direction, it applies only to *linear* edges, (and planar faces), as the docstring states.

> make the wires selectors as rich as the selectors of the faces and the edges?

The selector "<X[-2]" fails in your example with edges too.  The hole edges are not linear edges so the selector fails.

```py
from cadquery.occ_impl.shapes import *

edge1 = circle(1)
edge2 = circle(1).translate((1, 0, 0))
edge3 = circle(1).translate((2, 0, 0))
cmp_e = compound(edge1, edge2, edge3)

# this fails (if uncommented) - it doesn't matter if it is edge or wire
# sel_fail1 = cmp_e.edges(">X[-2]")

wire1 = wire(edge1)
wire2 = wire(edge2)
wire3 = wire(edge3)
cmp_w = compound(wire1, wire2, wire3)

# this also fails
# sel_fail2 = cmp_w.wires(">X[-2]")
```

`CenterNthSelector` was then added in #549.

```py
# CenterNthSelector works with all shapes
sel_ok = cmp_w.wires(">>X[-2]")
print(sel_ok.Center())
```

Here is a `DirectionNthSelector` example:

```py
from cadquery.occ_impl.shapes import *

edge1 = segment((0, 0), (10, 0))
edge2 = spline([(10, 0), (20, 20)], [(1, 0), (1, 0)])
edge3 = segment((20, 20), (30, 20))
wire1 = wire(edge1, edge2, edge3)

wire2 = rect(20, 20).translate((50, 10))

# there are 3 linear edges in  +X direction (DirectionSelector)
sel_x_edges = compound(wire1, wire2).edges("+X")
print(len(sel_x_edges.Edges()))

# selecting the middle edge in +X with DirectionNthSelector
sel_x_edge1 = compound(wire1, wire2).edges(">X[1]")
```

adam-urbanczyk, 2024-10-12:
@huskier original issue is resolved. You are not describing a clear new issue, please use discussions for general conversations.

closed 2024-10-12 by adam-urbanczyk

huskier, 2024-10-13:
Thank you for your explanation, @lorenzncode @adam-urbanczyk .