CadQuery/cadquery #1691: About the potential to enhance the assembly tree structure?

huskier, 2024-10-26:
The attached image is the screenshot of the door STEP model opened in FreeCAD, and the assembly tree structure is on the left. Is it possible to just keep the elements in the green rectangle box? To me, "door001", "Origin011", "Origin010", "left_part", "Origin001", "right_part", etc. are just duplications, and I think they are unnecessary.
[image]

adam-urbanczyk, 2024-10-26:
Is this based on CQ master, I remember there were some changes related to this. If so, would you know how FreeCAD maps elements from the XCAF structure?

huskier, 2024-10-27:
Yes, this is based on the latest CQ master (The "dfba42f" commit). We've also tried to open the STEP file in OnShape, and here is the result. We still just want to keep the elements in the green rectangle box. We've also tried to open the STEP file in SolidWorks, and the tree structure is more complex, not as expected as mentioned above. By "How does FreeCAD map elements from the XCAF structure", We're approaching the FreeCAD community, and no response right now.
[image]

huskier, 2024-10-28:
The following reply is from the FreeCAD community. So, we can manipulate the assemble tree structure. For how-to, we may need to investigate it in details.

> [Re: About the STEP model's tree structure](https://forum.freecad.org/viewtopic.php?p=789245#p789245)
> [Post](https://forum.freecad.org/viewtopic.php?p=789245#p789245) by [LHC](https://forum.freecad.org/memberlist.php?mode=viewprofile&u=36255) » Mon Oct 28, 2024 8:47 pm
> 
> The little blue box parts are the ones you want to keep. You can click on the other stuff and hit delete button, and do NOT delete the contents when it asks you. The benefit of the top level "part container" (the yellow box) is that you can move the entire group of items together as a unit, otherwise you will quickly find that they are all just positioned in space and can be moved around so they are not in the right place relative to each other. Here's a pro-tip - make a copy of the file - then start butchering the copy - don't ask me how I learned this :lol:

adam-urbanczyk, 2025-10-04:
Fixed in #1779

closed 2025-10-04 by adam-urbanczyk