CadQuery/cadquery #1695: Cutting a face with a solid

bragostin, 2024-10-30:
Before #1672 it was possible to perform a boolean cut of a face or shell with a solid. I guess it was not by design but because the test `isSolid` was not returning a correct value.
Now that `isSolid` is fixed it is not possible anymore since boolean operations require solids to work.
However it is actually quite useful to be able to cut a face with a solid. For example in additive manufacturing you often have surfaces instead of solids when you need very thin walled structures : it's the laser spot diameter that will determine the thickness of the wall and the CAD is much more lightweight with surfaces.

I managed to get back the previous behaviour by modifying `findSolid` like shown below.
Do you think it would make sense to extend boolean operations to surfaces permanently in CQ?

```python
    def findSolid(
        self, searchStack: bool = True, searchParents: bool = True
    ) -> Union[Solid, Compound]:
        """
        Finds the first solid object in the chain, searching from the current node
        backwards through parents until one is found.

        :param searchStack: should objects on the stack be searched first?
        :param searchParents: should parents be searched?
        :raises ValueError: if no solid is found
        
        This function is very important for chains that are modifying a single parent object,
        most often a solid.

        Most of the time, a chain defines or selects a solid, and then modifies it using workplanes
        or other operations.
    
        Plugin Developers should make use of this method to find the solid that should be modified,
        if the plugin implements a unary operation, or if the operation will automatically merge its
        results with an object already on the stack.
        """

        found = self._findType((Solid,), searchStack, searchParents)

        if found is None:
             found = self._findType((Compound,), searchStack, searchParents)
        if found is None:
             found = self._findType((Face,), searchStack, searchParents)

        if found is None:
            message = "on the stack or " if searchStack else ""
            raise ValueError(
                "Cannot find a solid {}in the parent chain".format(message)
            )

        return found
```

adam-urbanczyk, 2024-11-01:
It still possible to do that, just not with `cq.Workplane` directly.

bragostin, 2024-11-04:
Indeed, replacing
```python
object = object.cut(tool)
```
with
```python
object = cq.Workplane().newObject( [ cq.Shape.cut( object.val(), tool.val() ) ] )
```
allows to cut a surface with a solid.

closed 2024-11-04 by bragostin

adam-urbanczyk, 2024-11-04:
NB: on master you should be able to write it like so
```python
object = object.apply(lambda x: x - tool.val())
```

Also, you might want to simply consider switching to the free function API for this kind of use cases. See: https://cadquery.readthedocs.io/en/latest/free-func.html
```python
object: Shape = ...
tool: Shape = ...
object -= tool
```