CadQuery/cadquery #1697: Docs for overloads

adam-urbanczyk, 2024-11-04:
Currently oveload docstrings are simply ignored.

Possible solutions (TBC):
* https://sphinx-immaterial.readthedocs.io/en/latest/apidoc/python/apigen.html
* https://sphinx-toolbox.readthedocs.io/en/latest/extensions/more_autodoc/index.html

lorenzncode, 2024-11-24:
Current docs (to capture the current state):

`Workplane.split`
API Ref:
https://cadquery.readthedocs.io/en/latest/apireference.html
[image]

Class Summary:
[image]

`Assembly.add` is an example where the docstrings on the overloads are ignored.
[image]

The  `Workplane.split` overload docstrings are empty. docstrings added only to the non-@overload-decorated definition.

In the case of `Assembly.add` docstrings are added to the decorated and non-decorated defs.

lorenzncode, 2024-11-24:
I tried with `sphinx_toolbox.more_autodoc.overloads`.  It provides an on option `overloads_location`.

Here is `Assembly.add` with `overloads_location=signature`:

[image]

`overloads_location=top`
[image]

`overloads_location=bottom`
[image]

It didn't provide access to the other docstrings in the decorated defs, at least that I could find easily.  I'll take a look at the other possible solution.