CadQuery/cadquery #1703: FOSDEM 2025

adam-urbanczyk, 2024-11-16:
FOSDEM 2025 will have a CAD/open HW devroom:

https://lists.fosdem.org/pipermail/fosdem/2024q4/003569.html

I'm planning to submit an abstract about CQ and the new func API. Any other ideas?

jmwright, 2024-11-16:
@adam-urbanczyk It might be nice to have the community create a listing of some projects using CadQuery and turn that into a slide.

adam-urbanczyk, 2024-12-08:
The proposal was accepted.

adam-urbanczyk, 2024-12-15:
FYI: https://fosdem.org/2025/schedule/event/fosdem-2025-5286-free-function-api-for-cadquery/

jmwright, 2024-12-15:
Thanks Adam!

huskier, 2024-12-16:
I am very interested in the free function topic. With the free function, can CQ users build models like OpenSCAD users? Am I understanding it correctly?

jmwright, 2025-01-29:
Let's all make sure we support @adam-urbanczyk for this talk in whatever ways we can.

https://fosdem.org/2025/schedule/event/fosdem-2025-5286-free-function-api-for-cadquery/

michaelgale, 2025-01-31:
Looking forward to seeing it and learning about the new API scheme!

closed 2025-02-02 by adam-urbanczyk

jmwright, 2025-02-02:
@adam-urbanczyk Did this already happen? If so, I had my timezones wrong.

lorenzncode, 2025-02-02:
It was the second session in the morning today at 9:20 UTC+1.  Great talk and intro to the new API!

jmwright, 2025-02-02:
Sorry I messed it. I'll look forward to watching the video when it's posted. Thanks for doing this @adam-urbanczyk !

adam-urbanczyk, 2025-02-02:
NP, the  video should be there soon. Not my best talk, but the material is fine I think. I'll upload the sources.

adam-urbanczyk, 2025-02-02:
For the time being: [slides](https://fosdem.org/2025/events/attachments/fosdem-2025-5286-free-function-api-for-cadquery/slides/236108/slides_QLkQavI.html)

adam-urbanczyk, 2025-02-03:
https://github.com/CadQuery/conference-talks

jmwright, 2025-02-27:
I forgot to circle back around to this. You did a great job on the presentation @adam-urbanczyk . Thanks for your willingness to give that talk.