CadQuery/cadquery #1704: ImportError: cannot import name 'zoom' from 'ezdxf' (/opt/conda/lib/python3.12/site-packages/ezdxf/__init__.py)

pmbaumgartner, 2024-11-17:
On a fresh install (on macOS, using docker) I get:

```
ImportError: cannot import name 'zoom' from 'ezdxf' (/opt/conda/lib/python3.12/site-packages/ezdxf/__init__.py)
```

full traceback:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/conda/lib/python3.12/site-packages/cadquery/__init__.py", line 22, in <module>
    from .occ_impl import exporters
  File "/opt/conda/lib/python3.12/site-packages/cadquery/occ_impl/exporters/__init__.py", line 18, in <module>
    from .dxf import exportDXF, DxfDocument
  File "/opt/conda/lib/python3.12/site-packages/cadquery/occ_impl/exporters/dxf.py", line 6, in <module>
    from ezdxf import units, zoom
ImportError: cannot import name 'zoom' from 'ezdxf' (/opt/conda/lib/python3.12/site-packages/ezdxf/__init__.py)
```

kariem2k, 2024-12-30:
This is happening to me too on Windows (Clean Conda installation). I tried 2.4 and master both cause the same error.

jmwright, 2024-12-30:
Can you run `conda list` in your environment and post it here?

kariem2k, 2024-12-30:
Of course
[condalist.txt](https://github.com/user-attachments/files/18276275/condalist.txt)

Attached the file.

jmwright, 2025-01-01:
The version of ezdxf that output is showing is not even close to the current release. The oldest version I can find in my Linux environments is 1.1.0. I don't see any reason why conda would be installing an earlier version (a Windows package is available).

https://anaconda.org/conda-forge/ezdxf

Can you try to force the installation of 1.3.0 in your conda environment?

jmwright, 2025-01-07:
@pmbaumgartner @kariem2k Are either of you able to try the potential solution above?

jmwright, 2025-01-08:
@adam-urbanczyk Should we pin the version of ezdxf in environment.yml to force conda to use a current version? 

https://github.com/CadQuery/cadquery/blob/master/environment.yml#L16

jmwright, 2025-01-08:
@shimwell I have not seen evidence that pip is affected, have you? It seems like a conda solver issue. We could go ahead and add the pin to setup.py for completeness anyway.

jmwright, 2025-01-08:
@shimwell No worries. Some conda and pip packages are also maintained separately, so the versions do not always match up (one can lag the other). So that adds another layer of consideration to this. However, I do see that PyPI has version 1.3.0 of ezdxf and above, so we could add the pin for >=1.3.0 to setup.py as well to be more  consistent between environments.

jmwright, 2025-01-08:
I created #1746 to see if we can fix this.

jmwright, 2025-01-08:
#1746 has been merged. I think it usually takes a few hours for the packages to be available on conda-forge. Can someone who can reproduce this issue test a conda master install of CadQuery when the new packages are available? I believe you can watch for the new master package on conda-forge. When it is available there, you can try the install.

https://anaconda.org/CadQuery/cadquery/files

adam-urbanczyk, 2025-01-08:
Actually I need to update the recipe there.

adam-urbanczyk, 2025-01-09:
Done

closed 2025-01-09 by jmwright